nordicsemi/Android-BLE-Library #241: Getting a REASON_NULL_ATTRIBUTE on occasion

huntc, 2020-09-10:
On a reconnection, I observe that, on occasion, a subscription being attempted again i.e. where it has previously succeeded, fails:

```
D/au.com.titanclass.xdpservice.GattService$BleEndDeviceClient: gatt.setCharacteristicNotification(0e67ea13-b773-4699-bb66-202d4bbf40a7, true)
D/BluetoothGatt: setCharacteristicNotification() - uuid: 0e67ea13-b773-4699-bb66-202d4bbf40a7 enable: true
V/au.com.titanclass.xdpservice.GattService$BleEndDeviceClient: Enabling notifications for 0e67ea13-b773-4699-bb66-202d4bbf40a7
D/au.com.titanclass.xdpservice.GattService$BleEndDeviceClient: gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
E/au.com.titanclass.xdpservice.GattService$BleEndDeviceClient: Could not subscribe: -3
```

What to do in this scenario? Ignore it? Retry?

philips77, 2020-09-10:
Looks like 
```java
gatt.writeDescriptor(descriptor);
```
return `false`.

Is the descriptor reference up-to-date? This looks for me like a caching issue. You may try returning `true` from https://github.com/NordicSemiconductor/Android-BLE-Library/blob/1b836ad846ef86fed35fc93807ff8e1056afd57f/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L448
but this has other implications.

huntc, 2020-09-11:
> Is the descriptor reference up-to-date?

Yes, it should be as this is a re-connect scenario where a previous connection was successfully subscribed.

philips77, 2020-09-11:
What I meant was that it might have been nullified on disconnection (it's not, as uuid is logged) or not refreshed after the second service discovery (so it was obtained on the first connection), but I think even such should work, as all it's used is the handle number, which is the same.
I have never seen writeDescriptor or writeCharacteristic returning false. You could investigate that in the source code.

RaymondPelletier, 2020-09-21:
I am also seeing REASON_NULL_ATTRIBUTE on a regular basis with certain model phones. Samsung Note 10 are particularly bad. I can only connect once. Subsequent attempts result in  REASON_NULL_ATTRIBUTE errors when writing the characteristic.  To get the connection working requires terminating the app.

philips77, 2020-09-22:
Could you paste here some logs?

zetokz, 2021-01-20:
I can easily get this error in my app.
In my `initialize` function I have some code `readCharacteristic(authCharacteristic)` and the app waits for the result. Each time I call service discovery `initialize` function called and I see this REASON_NULL_ATTRIBUTE error. Besides the first initialize call where it works fine.

philips77, 2021-01-20:
Related to #273

philips77, 2022-03-21:
I'm closing the issue due to inactivity. Hopefully it does not recreate in the recent version. Feel free to create a new one otherwise.

closed 2022-03-21 by philips77