nordicsemi/Android-BLE-Library #242: SendNotification gets lost between two Android devices because of race condition in connection setup

jfkimmes, 2020-09-10:
I have set up two Android smartphones, one running Android-BLE-Library in client-mode, and one running Android-BLE-Library in server-mode.  

After scanning, when connecting, I use the `initialize()` method in the clients `BleManagerGattCallback` to set up a notification callback:

```java
setNotificationCallback(messageCharacteristic)
        .with(new DataReceivedCallback() {
            @Override
            public void onDataReceived(@NonNull BluetoothDevice device, @NonNull Data data) {
                // use data
            }
        });

enableNotifications(messageCharacteristic)
        .enqueue();
```

On the server side I use the `ConnectionObserver.onDeviceReady()` callback as a means to know when both devices are fully connected and ready to send/receive data, e.g like so:
```java
sendNotification(messageCharacteristic, new Data(data))
    .enqueue();
```

The problem is that the logs show that the server sends around 2 seconds before the client has set up its notification callback. If I delay the sending artificially everything works as expected.

Do I use the wrong callbacks to check the connection status or is this a bug?

jfkimmes, 2020-09-24:
Currently I simply "sleep" for 3 seconds before I continue working with the connection to be sure that the other side has finished setting up the connection. However, that is not really ideal, as my application is kind of time sensitive.

@philips77, could you comment on whether this is a bug or a fault on my end? (Sorry to bother you.)

philips77, 2020-09-24:
No, it's not a bug. It's just difficult to synchronize them. Both sides do service discovery of the other side and one of them always will be first. To be honest I didn't try that scenario, but it seems interesting. I wonder what happens with `enableNotifications` before the server is ready... I guess the delay is necessary here :(

jfkimmes, 2020-09-24:
If I understand correctly, this happens only because both devices act as a server **and** client and therefore both do the service discovery?
Personally I don't need them to act as both (in a PoC implementation I did before, this was *not* a problem since only one acted as a server and the other as a client. The vanilla android BLE API's expose a callback that is only called when the connection is indeed established), but what I got from #229 the library's inner structure doesn't really allow for this simpler setup, correct? 

Or is there a way for me to use the library only with one GATT server and the other only as a peripheral? If there's no such solution you can close this issue. 

Thanks alot for your assistance.

philips77, 2020-09-24:
> If I understand correctly, this happens only because both devices act as a server and client and therefore both do the service discovery?

There's #231 that can separate server only.

> the library's inner structure doesn't really allow for this simpler setup, correct?

Right. In the current form it does not. It was initially client-only, and I added server just recently.

I will go for 3 months leave very soon and will not be able to maintain the library for that time. I hope I'll merge some PRs, but can't implement this server-only setup as I would like it to be.

jfkimmes, 2020-09-24:
Alright, thanks again. :+1: 
I wish you a great time on your leave 😊️

closed 2020-09-24 by jfkimmes