nordicsemi/Android-BLE-Library #243: How do I connect to my Ble device?

andrew203, 2020-09-16:
When I finished the scan, I had the results. How do I connect to my Ble device?

   BluetoothLeScannerCompat scanner = BluetoothLeScannerCompat.getScanner();
        ScanSettings settings = new ScanSettings.Builder()
                .setLegacy(false)
                .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                .setReportDelay(1000)
                .setUseHardwareBatchingIfSupported(false)
                .build();
        List<ScanFilter> filters = new ArrayList<>();
      //  filters.add(new ScanFilter.Builder().setServiceUuid(mUuid).build());
        scanCallback=new ScanCallback() {
            @Override
            public void onScanResult(int callbackType, @NonNull ScanResult result) {
                super.onScanResult(callbackType, result);
            }

            @Override
            public void onBatchScanResults(@NonNull List<ScanResult> results) {
                super.onBatchScanResults(results);
                for(int i=0;i<results.size();i++){
                    Log.e(TAG,"onBatchScanResults="+results.get(i).getDevice().getAddress());
                }
            }

            @Override
            public void onScanFailed(int errorCode) {
                super.onScanFailed(errorCode);
            }
        };
        scanner.startScan(filters, settings, scanCallback);
        scanning = true;
    }

[image]

philips77, 2020-09-16:
Hi, have a look at our examples, e.g. here:
https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/2f3a06bf0e0c62a8172ac7fe2d872ac62ec19d84/app/src/main/java/no/nordicsemi/android/blinky/viewmodels/BlinkyViewModel.java#L66

closed 2020-09-16 by philips77

andrew203, 2020-09-17:
Thank you for your reply

andrew203, 2020-09-17:
> Hi, have a look at our examples, e.g. here:
> https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/2f3a06bf0e0c62a8172ac7fe2d872ac62ec19d84/app/src/main/java/no/nordicsemi/android/blinky/viewmodels/BlinkyViewModel.java#L66

Thank you for your reply

andrew203, 2020-09-17:
> https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/2f3a06bf0e0c62a8172ac7fe2d872ac62ec19d84/app/src/main/java/no/nordicsemi/android/blinky/viewmodels/BlinkyViewModel.java#L66

By the way，Besides the Blink project, what other examples can you refer to for connecting, sending and receiving data? I'm not familiar with LiveData

philips77, 2020-09-17:
Check out the 2 latest Pull Requests.