nordicsemi/Android-BLE-Library #244: BleManager   Question

andrew203, 2020-09-17:
I've seen some demo and there are many different types of BleManager such as BatteryManager BlinkManager.

So I want to ask, if my app wants to connect to a BLE device and then do a transceiver with that device. How do I define it

philips77, 2020-09-17:
You may find an example here:
https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/master/app/src/main/java/no/nordicsemi/android/blinky/profile/BlinkyManager.java

closed 2020-09-17 by philips77

andrew203, 2020-09-17:
> You may find an example here:
> https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/master/app/src/main/java/no/nordicsemi/android/blinky/profile/BlinkyManager.java

Can you set the length of the sending frame and the length of the receiving frame in this demo?

philips77, 2020-09-17:
The length of a packet is defined by MTU (actually MTU- 3 bytes). The default MTU is 23, which gives 20 bytes of payload.
You may request longer by using
```java
requestMtu(...).enqueue();
```
When you write longer packets without response, or send them with notifications/indications from a server, you may use `split()` method, or a variant with parameters:
```java
writeCharacteristic(..., data)
   .split()
   .enqueue();
```

andrew203, 2020-09-17:
> The length of a packet is defined by MTU (actually MTU- 3 bytes). The default MTU is 23, which gives 20 bytes of payload.
> You may request longer by using
>
> ```java
> requestMtu(...).enqueue();
> ``` [...]

Thank you for your reply