nordicsemi/Android-BLE-Library #245: run failed

andrew203, 2020-09-17:
Static interface methods are only supported starting with Android N (--min-api 24): float no.nordicsemi.android.ble.common.profile.bp.BloodPressureTypes.toKPa(float, int)

Default interface methods are only supported starting with Android N (--min-api 24): void no.nordicsemi.android.ble.common.profile.cgm.CGMSessionRunTimeCallback.onContinuousGlucoseMonitorSessionRunTimeReceivedWithCrcError(android.bluetooth.BluetoothDevice, no.nordicsemi.android.ble.data.Data)

Invoke-customs are only supported starting with Android O (--min-api 26)

Default interface methods are only supported starting with Android N (--min-api 24): void no.nordicsemi.android.ble.BleManagerCallbacks.onBatteryValueReceived(android.bluetooth.BluetoothDevice, int)

philips77, 2020-09-17:
Did you set Java 8 features in your app, like described here:
https://developer.android.com/studio/write/java8-support

andrew203, 2020-09-17:
> Did you set Java 8 features in your app, like described here:
> https://developer.android.com/studio/write/java8-support

[image]
I set in my project gradle

andrew203, 2020-09-17:
> Did you set Java 8 features in your app, like described here:
> https://developer.android.com/studio/write/java8-support
Thank you for your reply and I have solved it

closed 2020-09-17 by philips77