nordicsemi/Android-BLE-Library #250: autoConnect at the first connection

VincentMasselis, 2020-09-24:
Hey everyone,

I'm communicating with a BLE watch which automatically disable his bluetooth after 2 hours of disconnection, to communicate with the watch again, the user need to manually turn on the bluetooth on the watch. Because of this user interaction on the watch, the disconnection can last a very long time.

Based on this behavior my best scenario when the disconnection occurs was: 
* 0h to 1h59m of disconnection : _Scanning and trying to connect from the phone regularly_
* 2h to ? of disconnection : _Start using auto-connect to save battery life since the watch is probably not reachable at all_

In my case I'm trying to establish a connection as fast as possible without calling `useAutoConnect` and only after 2h of fails, I decide to change and switch to `useAutoConnect(true)` to save battery life. The problem is that Android-BLE-Library needs to establish a connection at least 1 time before using the flag `autoConnect` internally, unfortunately, that's where the troubles begin because I'm looking to use the flag `autoConnect` precisely because I don't need to connect quickly and I want to save battery life. But, by forcing the first connection without the flag `autoConnect` the connection will timeout in my scenario (even with `useAutoConnect(true)`) and no subsequents connections are started automatically with the `autoConnect` flag causing no battery life optimization at all.

My proposal is to transform the `autoConnect` boolean from `ConnectRequest` to something like:
```java
public enum AutoConnect {
   NEVER,
   AFTER_FIRST_CONNECTION,
   ALWAYS
}
```

Regards.

VincentMasselis, 2021-04-27:
Hey !
No prob for the late response.

Unfortunately, making a scan can consume a lot of battery and I still prefer the `autoConnect` solution. I'm made a lot of experiments on my 8.0 Nexus 5X which leads me to this:
* Worse case -> starting and stop scans even with a long pause like 30m. It trigger something like a wake lock which drains the battery life even if the phone is idle and screen off
* Normal case -> making a continuous scan with the LOW_POWER setting. It doesn't trigger any wake lock but it still affect the battery life compared to the phone without the running scan (I don't remember the numbers sorry...)
* Best case -> When using `autoConnect`, I was unable to see any difference regarding to the battery life between the phone running `autoConnect` and the phone which was not running the app. Less than 1% difference in 12h !

Of course, my result are valid only in my scenario on my phone, maybe a Huawei phone, with the dozen of embedded bloatware, could change the way my app interact with the system resulting to a better battery life.

philips77, 2021-04-27:
Did you try scanning with `PendingIntent`? On Android 8+ you start scan and forget. No background service required. You'll get a `PendingIntent` which can start your `BroadcastReceiver` from which you may connect and you know the device is in range, or at least close.

VincentMasselis, 2021-04-28:
Nope, I've never tested this way of scanning since I was targeting Android 6.0+ , maybe the battery life could be as good as `autoConnect`, I don't know..

philips77, 2022-03-21:
The related #226 has been fixed by #232. A connection to a bonded device initiated with `useAutoConnect(true)`, will retry on timeout with this flag set. This should fix your issue, and will also ensure quick connection when the device is in range.

closed 2022-03-21 by philips77