nordicsemi/Android-BLE-Library #251: Cannot cancel connection with using autoConnect

VincentMasselis, 2020-09-24:
I was connecting with `autoConnect` and, at some point, I decide to cancel the pending connection, the request connection is enqueued alone in the queue and nothing appends.

I'm using the last commit on develop (the one this morning), the cause of the bug seems of the bool `operationInProgress` set to true here ->
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/develop/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L2828 causing the disconnection request to be enqueued and but never launched.

philips77, 2020-09-24:
Yes... this use case may not be supported.

VincentMasselis, 2020-09-24:
😢 Do you have any workaround that I could use for the moment ?

philips77, 2020-09-24:
The initial connection, even with `autoConnect = true` is done with this param set to false (although see #250). It should timeout in 30 seconds, or when you set `.timeout(...)` parameter to the `ConnectRequest`. You may try calling `close()` and `connect(...).enqueue()` again, but I can't guarantee it will work as expected.

VincentMasselis, 2020-09-24:
By `close()` you mean the one from `BleManager` or `BleManagerHandler` ?

philips77, 2020-09-24:
True... How about using a different `BleManager` instance then?

VincentMasselis, 2020-09-24:
Firstly, i'm injecting my manager instance at some places in my code, it's not gonna be easy to swap from one instance to another because of this ioc.
Secondly, forgetting my previous `BleManager` instance doesn't stops the running `autoConnect` connection, which leads, i thinks, to memories leaks.

philips77, 2020-09-24:
You're right. Cancellation would really be a nice to have feature. I can look at it next week, but it's my last week before 3-month leave, so I can't promise much.

VincentMasselis, 2020-09-24:
I'm sending all the virtual encouragements I could behind my computer 💪

philips77, 2021-04-26:
I've started working on this feature now.

PR #292 (2021-04-26): Adding option to cancel pending connection 
This PR closes #251.

**Note:** It was possible to do the same before, using `BleManager#cancelQueue()`. The new method in `ConnectRequest` calls this method. Adding this method makes the API more readable.
closed 2021-04-26 by philips77