nordicsemi/Android-BLE-Library #253: LGE G8 ThinQ can't connect with `.usePreferredPhy(PHY_LE_2M_MASK)`

valeriyo, 2020-09-30:
I've been troubleshooting my app, not being able to connect on *LGE G8 ThinQ* running Android 9 (latest available OS from LG with all updates)

The symptoms were:
- `onDeviceConnecting` would arrive and after that -- nothing.
- LightBlue app was behaving same as my app - it would spin indefinitely when connecting.
- Turning Bluetooth OFF/ON didn't help.
- Restarting the phone didn't help.
- Clearing Bluetooth (system app) data didn't help.
- Factory resetting the phone didn't help.

After a while, I found out that:
- Right after reboot, LightBlue COULD connect. It would error out every other time (with "no data"), but still.
- After I tried with my app unsuccessfully, the LightBlue app COULD NOT connect as well.

Then, instead of reboot, I discovered that turning Airplane mode ON/OFF worked as well as restarting the phone.

Then, I started changing my app code. One thing I tried was remove `.usePreferredPhy(PHY_LE_2M_MASK)` from connection request, and that allowed my app to successfully connect (at the same rate as LightBlue).

Basically, conclusions are that on *LGE G8 ThinQ*:
1. `.usePreferredPhy(PHY_LE_2M_MASK)` hangs connection indefinitely, and
2. once `.usePreferredPhy(PHY_LE_2M_MASK)` was used -- it spoils all other connections as well, until phone is restarted, or Airplane Mode is turned ON/OFF.

My questions are - first of all, WTF is wrong with LG?!? And then, what's the best way to workaround this problem... remove this option completely, remove it for LGE only, dynamically detect this issue at runtime and remove this option, or something else?

Thanks!

P.S. Maybe this will help someone too, since I googled for a couple days, and found nothing useful.

valeriyo, 2020-09-30:
Next problem: WRITE either times out or fails with REASON_NULL_ATTRIBUTE.

philips77, 2021-04-27:
Thank for the info and sorry for such a long delay. I was busy with other tasks. 

We don't have any LG phone to test here. Did you try `.usePreferredPhy(PHY_LE_1M_MASK | PHY_LE_2M_MASK)`?

philips77, 2021-04-27:
Also, you may check [isLe2MPhySupported()](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#isLe2MPhySupported()) before setting LE 2M PHY.

closed 2021-04-27 by philips77