nordicsemi/Android-BLE-Library #255: BUG : onDeviceDisconnected is not called after a disconnection

pvarry, 2020-10-01:
I have a connection to a BLE device, when I disconnect this device programmatically, `onDeviceDisconnecting` is correctly call but `onDeviceDisconnected` was never called.
Library version: 2.2.4

The condition at this line is maybe the problem : https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L604
Following the comment 2 lines, the condition must be reversed.

This is a critical issue for us

philips77, 2020-10-01:
No, it looks correct. If the device was connected, there should be `onConnectionStateChanged` event after that, which will change the state to disconnected. In case it wasn't, I set the state manually.

pvarry, 2020-10-08:
For the comment on BleManagerHandler.java#L604, I got wrong and I don't know exactly how it work internally, excuse me for that comment.
It is normal to not receive the `onDeviceDisconnected` callback when the device was connected and the app request to close this connection ?

philips77, 2021-04-26:
Hello,
Sorry for coming back to you so late, but I was busy with other tasks. Did you manager to solve your problem?

philips77, 2021-08-31:
The callbacks were addressed in 2.3.0. The issue might have been fixed already and had some long time of inactivity, so I'll close it now. Please, reopen if needed, or create a new issue.

closed 2021-08-31 by philips77