nordicsemi/Android-BLE-Library #256: PAIRING_VARIANT_OOB_CONSENT question

J-ARTS, 2020-10-22:
Hello @philips77 

I have one question regarding PAIRING_VARIANT_OOB_CONSENT, do you know when it get's triggered currently I am try to use createBondOutOfBand using reflection and createBond(transport : Int) but this pairing variant does not get called.

I thought it will pass a OOB flag but it doesn't seems like that.

Do you know anything about how Android / Fluoride is handling it or maybe you know which specification where used for that

philips77, 2021-01-25:
Calling createBondOutOfBand should do the trick, it as far as I know. Which version on Android are you using?

J-ARTS, 2021-01-25:
Hi @philips77  thx for your response
I am testing it using two Android device with Android 10 and 11.
For reflection I am using https://github.com/ChickenHook/RestrictionBypass
because getting access to hidden api method are with the never Android versions much harder.

philips77, 2021-04-26:
Did you see: https://android.googlesource.com/platform/frameworks/base/+/96e6e4a0b383ee60ac2cd61fe55688d697f35574

philips77, 2022-03-21:
I'm closing the issue due to inactivity.