nordicsemi/Android-BLE-Library #257: Hello,

pedroSG94, 2020-10-22:


philips77, 2020-10-23:
0 means success. I guess you cancelled connection, which successfully got cancelled...

pedroSG94, 2020-10-23:
No, I didn't. For that reason it seems weird.
I only wait timeout and it return 0 instead of 10 as expected for timeout.
I will try share you a trace to know library flush.

ConnectGatt is called in this line but gattCallback never receive onConnectionStateChange:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L574
Internal connect is called from here:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L2947
Timeout of connect().fail() is called from here:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/TimeoutableRequest.java#L189
All ok here, but now internalDisconnect is called from here:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L1264  
But always return reason success (I think that in this case it should return Timeout):
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L613

Maybe I'm wrong about it.

philips77, 2021-04-26:
Sorry for coming so late to your issue, but I finally have time to work on BLE lib. I'll fix your issue. Expect it in version 2.3.

closed 2021-04-26 by philips77