nordicsemi/Android-BLE-Library #258: Ble devices are getting disconnected within 3-4 sec after connection.

pritam2102, 2020-11-03:
Hi,

We are  facing some issues between our ble devices (ble version 5) and android mobiles using nrfsdk ("Android-BLE-Library") library. (version 2.1.1)

We are able to connect our devices with mobile phones like samsung, redmi note 8 pro, redmi not 5 pro and other mobiles.

But when we are trying to connect same devices with mobiles like redmi note 8, redmi note 9 pro max and oppo A53. **it is getting disconnected within 3-4 sec with gatt timeout error 8.**

After tracing bluetooth snoop log we can see below error.
Attribute not found
{
13274 12783.807351 controller host HCI_EVT 8 Rcvd Number of Completed Packets
13275 12783.808066 remote () XiaomiCo_de:0a:37 (Redmi note 9 pro max) L2CAP 497 Rcvd Connection oriented channel
13276 12783.837010 Microchi_95:fc:5a (Endocare) XiaomiCo_de:0a:37 (Redmi note 9 pro max) ATT 74 Rcvd Read By Type Response, Attribute List Length: 3, Unknown, Unknown, Unknown
13277 12783.839225 XiaomiCo_de:0a:37 (Redmi note 9 pro max) Microchi_95:fc:5a (Endocare) ATT 16 Sent Read By Type Request, GATT Characteristic Declaration, Handles: 0x0057..0x0058
13278 12783.842390 controller host HCI_EVT 8 Rcvd Number of Completed Packets
13279 12783.858398 Microchi_95:fc:5a (Endocare) XiaomiCo_de:0a:37 (Redmi note 9 pro max) ATT 14 Rcvd Error Response - Attribute Not Found, Handle: 0x0057 (Unknown: Unknown)
}

We have already tried changing connection parameters like interval, latency and timeout but no luck.

We have attached screen shot of apk log file and hci snoop log file.
[image]
[image]

Please suggest on this.

philips77, 2020-11-03:
What's your min and max connection interval on the peripheral side?
Also, as I'm on leave until mid Jan, please ask your question on DevZone, so that the support team could help you.

Roshan0209, 2020-11-04:
Hello @philips77 , since my colleague (who has raised this issue) is on leave today, I will be responding to you. On the peripheral side the minimum connection interval is 10 ms, maximum interval is 30 ms, latency is 0  and timeout is 2000 ms.
For further reference, the module supports minimum and maximum connection interval of 7.5 ms to  4000 ms, latency of 0 to 499, and timeout of 100 ms to 32000 ms.
Also, it was my understanding that the connection parameters on the peripheral gets updated by the central, if so, then does changing these parameters on the peripheral side effect the connection?

philips77, 2020-11-04:
The central manages connection parameters, but in bound supported by the peripheral. Try increasing max interval on your peripheral to 45ms. I would say that the connection terminates as sides can't agree on that. But, as i said, consult with DevZone

Roshan0209, 2020-11-04:
Ok, Thank you for the reply. We will consult with DevZone as well. Will update here also once we conduct a full test after increasing  the maximum interval as well.

pritam2102, 2020-11-05:
We have tested the devices after increasing the max interval as suggested. Now the connection parameters of the peripheral are, min connection interval is 7.5 ms, max connection interval is 100 ms, latency is 0, and timeout is 5000 ms.
At first we tried normal connection, and were not able to connect. Then, we tried pairing our devices first and then tried connecting again, and were able to connect. Out of the 20 devices we tried we were able to connect to 19 of them after pairing. 1 device still disconnects after about 3 seconds after connection with an unknown error 14.

Is there any way to connect these devices without pairing them first? And any insight on why the devices connects only when paired will be helpful as well.

philips77, 2021-04-22:
Hello,
Sorry for very long delay. I was buried with other tasks. Did you manage to solve the connection issue?

philips77, 2022-03-21:
I'm closing the issue due to inactivity. Should you still experience similar problem, please reopen it or create a new issue.

closed 2022-03-21 by philips77

deeds-29, 2022-07-11:
hi, I am facing this issue in OPPO A5s Android version 8.1.0 colorOS version V5.2.1 BLE version 4.2