nordicsemi/Android-BLE-Library #260: Link layer PDU size larger than 27 bytes?

astenman, 2020-11-04:
I have a use case where an Arduino Nano BLE (using a custom app and Mbed OS 6.2) sends data to a custom Android app using Coded PHY. The data rate is approximately 200 bytes at 15 Hz. The Android app is running on a Motorola Edge phone (Android 10) which implements Bluetooth 5.1

The problem is that link layer PDU size is always selected as 27 bytes during the negotiation. Both devices should support the data length extension.

I tried the native Bluetooth LE API in Android first. But I get the same problem using this library. Is it possible to utilize the link layer PDU size of 251 bytes instead of as now 27 bytes?

Since it is not possible to sniff the on-air traffic using the nRF sniffer, I have used Sniffle (https://github.com/nccgroup/Sniffle) instead. 

During the negotiation, the slave (Arduino) sends an LL_LENGTH_REQ with RX and TX size 251, but the master (the phone) answers with TX size 27.

```
Timestamp: 4.810122	Length: 11	RSSI: -44	Channel: 24	PHY: Coded
LLID: LL CONTROL
Dir: S->M NESN: 1 SN: 0 MD: 0 Data Length: 9
Opcode: LL_LENGTH_REQ
07 09 14 FB 00 50 08 FB 00 48 08
               P           H   

Timestamp: 4.853677	Length: 11	RSSI: -35	Channel: 20	PHY: Coded
LLID: LL CONTROL
Dir: M->S NESN: 1 SN: 1 MD: 0 Data Length: 9
Opcode: LL_LENGTH_RSP
0F 09 15 FB 00 90 42 1B 00 90 0A
                  B            
```

During the data transmission the link layer PDU size is clearly 27 bytes as show by the following log

```
Timestamp: 7.118446	Length: 29	RSSI: -44	Channel: 24	PHY: Coded
LLID: LL DATA CONT
Dir: S->M NESN: 0 SN: 1 MD: 1 Data Length: 27
19 1B 3F 2D 00 40 9C 3E 35 00 00 7A 3F 3D 00 40 1C BF 45 00 00 90 41 4D 00 00 00 00 55
      ?  -     @     >  5        z  ?  =     @        E           A  M              U

Timestamp: 7.121044	Length: 2	RSSI: -35	Channel: 24	PHY: Coded
LLID: LL DATA CONT
Dir: M->S NESN: 0 SN: 0 MD: 0 Data Length: 0
01 00
    

Timestamp: 7.121914	Length: 29	RSSI: -44	Channel: 24	PHY: Coded
LLID: LL DATA CONT
Dir: S->M NESN: 1 SN: 0 MD: 1 Data Length: 27
15 1B 00 00 00 00 0A 32 08 F7 E0 F3 02 15 00 00 97 3C 1D 00 00 78 3B 25 00 90 7B 3F 2D
                     2                             <           x  ;  %        {  ?  -

Timestamp: 7.124512	Length: 2	RSSI: -35	Channel: 24	PHY: Coded
LLID: LL DATA CONT
Dir: M->S NESN: 1 SN: 1 MD: 0 Data Length: 0
0D 00
```

Is it possible to increase the PDU size?

astenman, 2020-11-05:
I can add that I set ATT MTU to 512 with success ...

```
        override fun initialize() {
            beginAtomicRequestQueue()
                .add(requestMtu(512)
                    .with { _, mtu -> Timber.i("MTU set to $mtu") }
                    .fail { _, status -> Timber.e("Requested MTU not supported: $status")})
                .add(setPreferredPhy(phy, phy, phyOptions)
                    .with { device, txPhy, _ -> broadcastUpdatePhy(device.address, txPhy) }
                    .fail { _, status -> Timber.e("Requested PHY not supported: $status")})
                .add(enableNotifications(imuCharacteristic))
                .done { Timber.i("Target initialized") }
                .enqueue()
            writeCharacteristic(imuCharacteristic, byteArrayOf(1))
                .done { Timber.i("Start command sent") }
                .enqueue()

            setNotificationCallback(imuCharacteristic).with(imuNotificationCallback)
        }
```

zetokz, 2021-01-20:
@astenman try to make sure DLE is enabled on the Bluetooth device.
Once you are confident it is enabled try to open L2CAP channel and send some data to see the PDU size.
`BluetoothDevice.createL2capChannel()`

From my experience, I can say that PDU started being 251 once we enabled DLE on the peripheral.

philips77, 2021-04-22:
Hello @astenman,
Yes, @zetokz is right. You need to make sure that DLE is enabled on the device side. With requesting higher MTU on Android side it will also use the DLE as you wanted if only it's available on both sides.

closed 2021-04-22 by philips77