nordicsemi/Android-BLE-Library #263: NullPointerException

979451341, 2020-11-23:
no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.lambda$nextRequest$3$BleManager$BleManagerGattCallback(BleManager.java:3408)

no.nordicsemi.android:ble:2.1.1

roshanrajaratnam, 2020-11-23:
Hi could you post some more information with logs please?

979451341, 2020-11-23:
> Hi could you post some more information with logs please?

1 no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.lambda$nextRequest$3$BleManager$BleManagerGattCallback(BleManager.java:3408)
--
2 no.nordicsemi.android.ble.-$$Lambda$BleManager$BleManagerGattCallback$lHohduzEoh65ydpbEB2__CX1LGU.run(Unknown Source:4)
3 android.os.Handler.handleCallback(Handler.java:883)
4 android.os.Handler.dispatchMessage(Handler.java:100)
5 android.os.Looper.loop(Looper.java:224)
6 android.app.ActivityThread.main(ActivityThread.java:7584)
7 java.lang.reflect.Method.invoke(Native Method)
8 com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)
9 com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)

philips77, 2021-04-22:
This could have been fixed in the latest releases. Please migrate to 2.2.4.
I'll be releasing 2.3 in few days with better Kotlin support.

closed 2021-04-22 by philips77