nordicsemi/Android-BLE-Library #266: Does not receive data after the device was connected

Stalker11, 2020-12-18:
Hello
I have some issues. Method on `onDataReceived` does not call after the device was connected (method `onDeviceConnected` was called). But for another device, it works fine.
These issues reproduce on v2.2.4. Library version 1.2.0 works fine.
What does it depend on?
Could you help me solve this problem?
Code:
[image]

zetokz, 2021-01-20:
@Stalker11 could you, please, attach some logs from your device to the issue?
It is almost impossible to say what is wrong, from a code perspective at least what is visible here everything is fine =)

philips77, 2021-04-22:
Hello @Stalker11,
Did you manage to solve the issue? Sorry for coming back to you so late, I was busy with other tasks.

From the top of my mind, the callback may not getting called for the following reasons:
1. your characteristics are *null*
2. they don't have READ property

In both cases you should get `fail` callback instead.

philips77, 2022-03-21:
Closed due to inactivity.

closed 2022-03-21 by philips77