nordicsemi/Android-BLE-Library #267: Are there any considerations for setting the access permissions of commonly used methods to protected?

fuhaodev, 2020-12-22:
Below is my code for connecting multiple devices. At that time, BleClient could not access the protected writeCharacteristic method. The current practice is to override the methods that need to be used and modify the access permissions to public, but this is not very elegant.

- BleClientManager

```
public class BleClientManager {

    private static final String TAG = BleClientManager.class.getSimpleName();
    private Map<String, BleClient> mClientMap = new HashMap<>();
    private BluetoothGattCharacteristic mBluetoothGattCharacteristic;

    private static final class Holder {
        private static final BleClientManager INSTANCE = new BleClientManager();
    }

    public static BleClientManager getInstance() {
        return Holder.INSTANCE;
    }

    private BleClient getBleClient(String mac) {
        BleClient client = mClientMap.get(mac);
        if (client == null) {
            client = new BleClient(BluetoothContext.get());
            mClientMap.put(mac, client);
        }
        return client;
    }

    public void connect(@NonNull final BluetoothDevice device) {
        BleClient client = getBleClient(device.getAddress());
        client.setConnectionObserver(this);
        client.connect(device)
                .useAutoConnect(true)
                .timeout(10000)
                .retry(5, 200)
                .done(device1 -> Logger.d(TAG, "Device initiated, " + device1.toString()));
    }

    public void write(String mac, byte[] data) {
        getBleClient(mac).writeCharacteristic();
    }

    public LiveData<ConnectionState> getState(String mac) {
        BleClient client = mClientMap.get(mac);
        if (client == null) {
            return null;
        }
        return client.getState();
    }
}
```

- BleClient

```
public class BleClient extends ObservableBleManager {

    private static final String TAG = BleClient.class.getSimpleName();

    ...
    ...

    @NonNull
    @Override
    public WriteRequest writeCharacteristic(@Nullable BluetoothGattCharacteristic characteristic, @Nullable Data data) {
        return super.writeCharacteristic(characteristic, data);
    }
}
```

okbean, 2020-12-23:
strongly suggest, only one device is in connecting process , when connecting multiple devices。in this case，your code creating ble client for every device would case connecting many device at the same time.t

philips77, 2021-04-22:
The reason why those methods are protected is that they are not intended to be exposed as device API.
That's why you have the `BleManager` class, to make it a layer between the BLE (transport) implementation and device API (application layer).
For example, your manager may have methods like `sendCommand(...)`, `reset()`, `enableWarpDrive(warpLevel)`, but the implementation should be hidden to the manager. Of course you may expose `writeCharacteristic`, but then you also expose characteristics, etc. Not needed.

In your case, the `BleClientManager` should not have `mBluetoothGattCharacteristic`, as each client has their own characteristics. The `BleClient` API should just be:
```java
public void write(String mac, byte[] data) {
        getBleClient(mac).send(data);
    }
```
BleClient:
```java
public class BleClient extends ObservableBleManager {
    private BluetoothGattCharacteristic mBluetoothGattCharacteristic;

    private static final String TAG = BleClient.class.getSimpleName();

    [...]

    @NonNull
    @Override
    public WriteRequest send(@Nullable Data data) {
        return super.writeCharacteristic(mBluetoothGattCharacteristic, data);
    }
}
```

philips77, 2021-04-22:
Also, sorry it took so long for me to reply to you, I was busy with other tasks.

closed 2021-04-22 by philips77