nordicsemi/Android-BLE-Library #268: initialize() was called twice

jiajunhui, 2020-12-26:
I'm using version 2.2.4
If the device is already bound, initialization will be called multiple times.
I found that when the device is in the bound state, the BleManagerHandler initiates a discovery service again
```
case BluetoothDevice.BOND_BONDED:
	log(Log.INFO, "Device bonded");
	postCallback(c -> c.onBonded(device));
	postBondingStateChange(o -> o.onBonded(device));
	if (request != null && request.type == Request.Type.CREATE_BOND) {
		request.notifySuccess(device);
		request = null;
		break;
	}
	// If the device started to pair just after the connection was
	// established the services were not discovered.
	if (!servicesDiscovered && !serviceDiscoveryRequested) {
		serviceDiscoveryRequested = true;
		post(() -> {
			log(Log.VERBOSE, "Discovering services...");
			log(Log.DEBUG, "gatt.discoverServices()");
			bluetoothGatt.discoverServices();
		});
		return;
	}
```

jiajunhui, 2020-12-29:
On Samsung's S10 device, the initialization method is called repeatedly instead of twice

philips77, 2021-01-19:
Thank you for creating this issue. Could you paste the logs here?

philips77, 2021-01-19:
Related to #120.

philips77, 2021-04-21:
This might have been fixed by 65e08d5.

philips77, 2021-04-22:
Perhaps fixed internally and will be released as 2.3. Closing this issue for now, please reopen if 2.3 still fails.

closed 2021-04-22 by philips77