nordicsemi/Android-BLE-Library #271: Failed to connect ble & setCharacteristicNotification does not request

NamSSu, 2021-01-12:
Hello! I am making a BLE communication application using this library. application is use this library version 2.2.4

During a recent test, I confirmed that it does not work well on the Galaxy S7 device, and it works well on the S8, S9, S10 and old S6.

My source code and connect source are as follows.

__BleManager__

```java
class MyBleManager extends BleManager {
    final static UUID SERVICE_UUID = UUID.fromString("6e400001-b5a3-f393-e0a9-e50e24dcca9e");
    final static UUID FIRST_CHAR   = UUID.fromString("6e400003-b5a3-f393-e0a9-e50e24dcca9e");

    // Client characteristics
    private BluetoothGattCharacteristic firstCharacteristic;

    MyBleManager(@NonNull final Context context) {
        super(context);
    }

    @NonNull
    @Override
    protected BleManagerGattCallback getGattCallback() {
        return new MyManagerGattCallback();
    }

    @Override
    public void log(final int priority, @NonNull final String message) {
        if (priority == Log.ERROR) {
            Log.println(priority, "MyBleManager", message);
        }
    }

    /**
     * BluetoothGatt callbacks object.
     */
    private class MyManagerGattCallback extends BleManagerGattCallback {
        @Override
        public boolean isRequiredServiceSupported(@NonNull final BluetoothGatt gatt) {
            final BluetoothGattService service = gatt.getService(SERVICE_UUID);
            if (service != null) {
                firstCharacteristic = service.getCharacteristic(FIRST_CHAR);

            }

            // Validate properties
            boolean notify = false;
            if (firstCharacteristic != null) {
                final int properties = firstCharacteristic.getProperties();
                notify = (properties & BluetoothGattCharacteristic.PROPERTY_NOTIFY) != 0;
            }
            
            return firstCharacteristic != null && notify;
        }

        // If you have any optional services, allocate them here. Return true only if
        // they are found.
        @Override
        protected boolean isOptionalServiceSupported(@NonNull final BluetoothGatt gatt) {
            return super.isOptionalServiceSupported(gatt);
        }

        // Initialize your device here. Often you need to enable notifications and set required
        // MTU or write some initial data. Do it here.
        @Override
        protected void initialize() {
            // You may enqueue multiple operations. A queue ensures that all operations are
            // performed one after another, but it is not required.
            beginAtomicRequestQueue()
                    .add(requestMtu(247) // Remember, GATT needs 3 bytes extra. This will allow packet size of 244 bytes.
                            .with((device, mtu) -> log(Log.INFO, "MTU set to " + mtu))
                            .fail((device, status) -> log(Log.WARN, "Requested MTU not supported: " + status)))
                    .add(setPreferredPhy(PhyRequest.PHY_LE_2M_MASK, PhyRequest.PHY_LE_2M_MASK, PhyRequest.PHY_OPTION_NO_PREFERRED)
                            .fail((device, status) -> log(Log.WARN, "Requested PHY not supported: " + status)))
                    .add(enableNotifications(firstCharacteristic))
                    .done(device -> log(Log.INFO, "Target initialized"))
                    .enqueue();
            
            setNotificationCallback(firstCharacteristic).with(new DataReceivedCallback() {
                @Override
                public void onDataReceived(@NonNull BluetoothDevice device, @NonNull Data data) {
                    Log.d("MyBleManager", data.getStringValue(0));
                }
            });

        }

        @Override
        protected void onDeviceDisconnected() {
            // Device disconnected. Release your references here.
            firstCharacteristic = null;
        }
    }

    public void abort() {
        cancelQueue();
    }
}
```

__Connect__

```java
        manager = new MyBleManager(this);
        manager.connect(d)
                .timeout(100000)
                .retry(3, 100)
                .done(device -> Log.i("MainActivity", "Device initiated"))
                .enqueue();
```

Logcat of Galaxy S7 device(Android 8.0.0) cannot be connected is as follows.
```
D/BluetoothGatt: connect() - device: EA:C4:26:58:9A:10, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=8ffbc805-e879-44fc-8e7c-1b2e9aee82d3
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=EA:C4:26:58:9A:10
D/BluetoothGatt: discoverServices() - device: EA:C4:26:58:9A:10
D/BluetoothGatt: onConnectionUpdated() - Device=EA:C4:26:58:9A:10 interval=24 latency=0 timeout=200 status=0
D/BluetoothGatt: onConnectionUpdated() - Device=EA:C4:26:58:9A:10 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt: onSearchComplete() = Device=EA:C4:26:58:9A:10 Status=0
D/BluetoothGatt: configureMTU() - device: EA:C4:26:58:9A:10 mtu: 247
D/BluetoothGatt: onConfigureMTU() - Device=EA:C4:26:58:9A:10 mtu=23 status=0
D/BluetoothGatt: onConnectionUpdated() - Device=EA:C4:26:58:9A:10 interval=24 latency=0 timeout=200 status=0
```
Logcat of the connected S6 device(Android 7.0.0) is as follows.
```
D/BluetoothGatt: connect() - device: EA:C4:26:58:9A:XX, auto: false
D/BluetoothAdapter: isSecureModeEnabled
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=ca7975ae-4b23-40ee-8df7-a278bfd75c24
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=EA:C4:26:58:9A:XX
D/BluetoothGatt: discoverServices() - device: EA:C4:26:58:9A:XX
D/BluetoothGatt: onClientConnParamsChanged() - Device=EA:C4:26:58:9A:XX interval=24 status=0
D/BluetoothGatt: onClientConnParamsChanged() - Device=EA:C4:26:58:9A:XX interval=6 status=0
D/BluetoothGatt: onSearchComplete() = Device=EA:C4:26:58:9A:XX Status=0
D/BluetoothGatt: configureMTU() - device: EA:C4:26:58:9A:XX mtu: 247
D/BluetoothGatt: onClientConnParamsChanged() - Device=EA:C4:26:58:9A:XX interval=24 status=0
D/BluetoothGatt: onConfigureMTU() - Device=EA:C4:26:58:9A:XX mtu=23 status=0
D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
```

In my opinion, setCharacteristicNotification cannot be executed, so it doesn't seem to be recevie.

Please let me know if there is something I did wrong or missed.
Thank you.

zetokz, 2021-01-20:
Could you please print all logs here
```java
    @Override
    public void log(final int priority, @NonNull final String message) {
        if (priority == Log.ERROR) {
            Log.println(priority, "MyBleManager", message);
        }
    }
```
and paste them here?

> Hello! I am making a BLE communication application using this library. application is use this library version 2.2.4
>
> During a recent test, I confirmed that it does not work well on the Galaxy S7 [...]

NamSSu, 2021-01-21:
@zetokz 
Thanks for the answer!

If priority == Log.ERROR is applied, no message is displayed on the device in question(Galaxy S7), so I changed the code as follows and saved the log.
```java
    @Override
    public void log(final int priority, @NonNull final String message) {
        Log.println(priority, "MyBleManager", message);
    }
```

log of Galaxy S6 Device(Android 7.0.0) is as follow.
```
2021-01-21 11:15:28.247 14948-14948/com.namysu.bletest V/MyBleManager: Connecting...
2021-01-21 11:15:28.247 14948-14948/com.namysu.bletest D/MyBleManager: gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE)
2021-01-21 11:15:28.559 14948-14962/com.namysu.bletest D/MyBleManager: [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2021-01-21 11:15:28.559 14948-14962/com.namysu.bletest I/MyBleManager: Connected to EA:C4:26:58:9A:10
2021-01-21 11:15:28.560 14948-14962/com.namysu.bletest D/MyBleManager: wait(300)
2021-01-21 11:15:28.863 14948-14948/com.namysu.bletest V/MyBleManager: Discovering services...
2021-01-21 11:15:28.863 14948-14948/com.namysu.bletest D/MyBleManager: gatt.discoverServices()
2021-01-21 11:15:29.425 14948-14962/com.namysu.bletest I/MyBleManager: Services discovered
2021-01-21 11:15:29.425 14948-14962/com.namysu.bletest V/MyBleManager: Primary service found
2021-01-21 11:15:29.426 14948-14962/com.namysu.bletest V/MyBleManager: Requesting new MTU...
2021-01-21 11:15:29.426 14948-14962/com.namysu.bletest D/MyBleManager: gatt.requestMtu(247)
2021-01-21 11:15:29.494 14948-15036/com.namysu.bletest I/MyBleManager: MTU changed to: 23
2021-01-21 11:15:29.494 14948-14948/com.namysu.bletest I/MyBleManager: MTU set to 23
2021-01-21 11:15:29.495 14948-15036/com.namysu.bletest D/MyBleManager: gatt.setCharacteristicNotification(6e400003-b5a3-f393-e0a9-e50e24dcca9e, true)
2021-01-21 11:15:29.496 14948-15036/com.namysu.bletest V/MyBleManager: Enabling notifications for 6e400003-b5a3-f393-e0a9-e50e24dcca9e
2021-01-21 11:15:29.496 14948-15036/com.namysu.bletest D/MyBleManager: gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
2021-01-21 11:15:29.553 14948-14961/com.namysu.bletest I/MyBleManager: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
2021-01-21 11:15:29.553 14948-14961/com.namysu.bletest I/MyBleManager: Notifications enabled
2021-01-21 11:15:29.554 14948-14948/com.namysu.bletest I/MyBleManager: Target initialized
2021-01-21 11:15:31.474 14948-14962/com.namysu.bletest I/MyBleManager: Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 76-65-72-3A-31-2E-30-2E-30
2021-01-21 11:15:31.475 14948-14948/com.namysu.bletest D/MyBleManager: ver:1.0.0
2021-01-21 11:15:32.289 14948-15036/com.namysu.bletest I/MyBleManager: Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 32-32-2E-31-38-23-33-35-39
2021-01-21 11:15:32.291 14948-14948/com.namysu.bletest D/MyBleManager: 22.18#359
2021-01-21 11:15:37.300 14948-14961/com.namysu.bletest I/MyBleManager: Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 32-32-2E-32-37-23-33-35-39
2021-01-21 11:15:37.301 14948-14948/com.namysu.bletest D/MyBleManager: 22.27#359
```

log of Galaxy S7 Device(Android 8.0.0) is as follow.
```
2021-01-21 11:09:49.573 3546-3546/com.namysu.bletest V/MyBleManager: Connecting...
2021-01-21 11:09:49.574 3546-3546/com.namysu.bletest D/MyBleManager: gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2021-01-21 11:09:51.258 3546-3597/com.namysu.bletest D/MyBleManager: [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2021-01-21 11:09:51.259 3546-3597/com.namysu.bletest I/MyBleManager: Connected to EA:C4:26:58:9A:10
2021-01-21 11:09:51.263 3546-3597/com.namysu.bletest D/MyBleManager: wait(300)
2021-01-21 11:09:51.566 3546-3546/com.namysu.bletest V/MyBleManager: Discovering services...
2021-01-21 11:09:51.566 3546-3546/com.namysu.bletest D/MyBleManager: gatt.discoverServices()
2021-01-21 11:09:52.020 3546-3597/com.namysu.bletest I/MyBleManager: Connection parameters updated (interval: 30.0ms, latency: 0, timeout: 2000ms)
2021-01-21 11:09:52.294 3546-3597/com.namysu.bletest I/MyBleManager: Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
2021-01-21 11:09:52.389 3546-3597/com.namysu.bletest I/MyBleManager: Services discovered
2021-01-21 11:09:52.390 3546-3597/com.namysu.bletest V/MyBleManager: Primary service found
2021-01-21 11:09:52.405 3546-3597/com.namysu.bletest V/MyBleManager: Requesting new MTU...
2021-01-21 11:09:52.405 3546-3597/com.namysu.bletest D/MyBleManager: gatt.requestMtu(247)
2021-01-21 11:09:52.424 3546-3597/com.namysu.bletest I/MyBleManager: Connection parameters updated (interval: 30.0ms, latency: 0, timeout: 2000ms)
2021-01-21 11:09:52.455 3546-3597/com.namysu.bletest I/MyBleManager: MTU changed to: 23
2021-01-21 11:09:52.457 3546-3546/com.namysu.bletest I/MyBleManager: MTU set to 23
2021-01-21 11:09:52.458 3546-3597/com.namysu.bletest V/MyBleManager: Requesting preferred PHYs...
2021-01-21 11:09:52.459 3546-3597/com.namysu.bletest D/MyBleManager: gatt.setPreferredPhy(LE 2M, LE 2M, coding option = No preferred)
2021-01-21 11:11:29.622 3546-3546/com.namysu.bletest V/MyBleManager: Disconnecting...
2021-01-21 11:11:29.624 3546-3546/com.namysu.bletest D/MyBleManager: gatt.disconnect()
[... cut ...]
2021-01-21 11:16:53.284 16675-16879/com.namysu.bletest I/MyBleManager: Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 32-33-2E-32-31-23-33-36-31
2021-01-21 11:16:53.285 16675-16675/com.namysu.bletest D/MyBleManager: 23.21#361
2021-01-21 11:16:58.295 16675-16879/com.namysu.bletest I/MyBleManager: Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 32-33-2E-32-34-23-33-36-30
2021-01-21 11:16:58.297 16675-16675/com.namysu.bletest D/MyBleManager: 23.24#360
```

zetokz, 2021-01-21:
@NamSSu thanks for the logs. It is much better now
But what I see from the new logs that the device now is disconnected after PHY request what I have not noticed in the logs you attached at the top.

As an option could you try to remove this PHY request and see what happens? 
And if possible attach result logs once you remove PHY request but without filtering by `MyBleManager` tag just filter by the application. We would see what is going on on all the levels.

NamSSu, 2021-01-25:
I don't know what happened, but after I removed the PHY it started magically connecting.
```java
        @Override
        protected void initialize() {
            // You may enqueue multiple operations. A queue ensures that all operations are
            // performed one after another, but it is not required.
            beginAtomicRequestQueue()
                    .add(requestMtu(247) // Remember, GATT needs 3 bytes extra. This will allow packet size of 244 bytes.
                            .with((device, mtu) -> log(Log.INFO, "MTU set to " + mtu))
                            .fail((device, status) -> log(Log.WARN, "Requested MTU not supported: " + status)))
/*                    .add(setPreferredPhy(PhyRequest.PHY_LE_2M_MASK, PhyRequest.PHY_LE_2M_MASK, PhyRequest.PHY_OPTION_NO_PREFERRED)
                            .fail((device, status) -> log(Log.WARN, "Requested PHY not supported: " + status)))*/ // disable phy
                    .add(enableNotifications(firstCharacteristic))
                    .done(device -> log(Log.INFO, "Target initialized"))
                    .enqueue();

            setNotificationCallback(firstCharacteristic).with(new DataReceivedCallback() {
                @Override
                public void onDataReceived(@NonNull BluetoothDevice device, @NonNull Data data) {
                    Log.d("MyBleManager", data.getStringValue(0));
                }
            });

        }
```

Below is the log of S6.
```
2021-01-25 10:19:54.218 20411-20411/com.namysu.bletest V/MyBleManager: Connecting...
2021-01-25 10:19:54.218 20411-20411/com.namysu.bletest D/MyBleManager: gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE)
2021-01-25 10:19:54.222 20411-20411/com.namysu.bletest D/BluetoothGatt: connect() - device: EA:C4:26:58:9A:XX, auto: false
2021-01-25 10:19:54.223 20411-20411/com.namysu.bletest D/BluetoothAdapter: isSecureModeEnabled
2021-01-25 10:19:54.223 20411-20411/com.namysu.bletest D/BluetoothGatt: registerApp()
2021-01-25 10:19:54.223 20411-20411/com.namysu.bletest D/BluetoothGatt: registerApp() - UUID=2e76a637-ff46-449f-ac99-055f3a6f740b
2021-01-25 10:19:54.328 20411-20425/com.namysu.bletest D/BluetoothGatt: onClientRegistered() - status=0 clientIf=4
2021-01-25 10:19:54.346 20411-20411/com.namysu.bletest D/ViewRootImpl@e34f5bd[MainActivity]: ThreadedRenderer.create() translucent=false
2021-01-25 10:19:54.349 20411-20411/com.namysu.bletest D/InputTransport: Input channel constructed: fd=57
2021-01-25 10:19:54.350 20411-20411/com.namysu.bletest D/ViewRootImpl@e34f5bd[MainActivity]: setView = DecorView@c9a67b2[MainActivity] touchMode=true
2021-01-25 10:19:54.360 20411-20411/com.namysu.bletest D/ViewRootImpl@e34f5bd[MainActivity]: dispatchAttachedToWindow
2021-01-25 10:19:54.371 20411-20411/com.namysu.bletest D/ViewRootImpl@e34f5bd[MainActivity]: Relayout returned: oldFrame=[0,0][0,0] newFrame=[0,0][1440,2560] result=0x7 surface={isValid=true 506879911424} surfaceGenerationChanged=true
2021-01-25 10:19:54.371 20411-20411/com.namysu.bletest D/ViewRootImpl@e34f5bd[MainActivity]: mHardwareRenderer.initialize() mSurface={isValid=true 506879911424} hwInitialized=true
2021-01-25 10:19:54.383 20411-20487/com.namysu.bletest D/libEGL: loaded /vendor/lib64/egl/libGLES_mali.so
2021-01-25 10:19:54.392 20411-20411/com.namysu.bletest D/ViewRootImpl@e34f5bd[MainActivity]: MSG_RESIZED_REPORT: ci=Rect(0, 96 - 0, 0) vi=Rect(0, 96 - 0, 0) or=1
2021-01-25 10:19:54.392 20411-20411/com.namysu.bletest D/ViewRootImpl@e34f5bd[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 1
2021-01-25 10:19:54.392 20411-20411/com.namysu.bletest D/ViewRootImpl@e34f5bd[MainActivity]: mHardwareRenderer.initializeIfNeeded()#2 mSurface={isValid=true 506879911424}
2021-01-25 10:19:54.393 20411-20411/com.namysu.bletest V/InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@8915c5f nm : com.namysu.bletest ic=null
2021-01-25 10:19:54.393 20411-20411/com.namysu.bletest I/InputMethodManager: [IMM] startInputInner - mService.startInputOrWindowGainedFocus
2021-01-25 10:19:54.397 20411-20424/com.namysu.bletest D/InputTransport: Input channel constructed: fd=58
2021-01-25 10:19:54.408 20411-20487/com.namysu.bletest I/OpenGLRenderer: Initialized EGL, version 1.4
2021-01-25 10:19:54.408 20411-20487/com.namysu.bletest D/OpenGLRenderer: Swap behavior 1
2021-01-25 10:19:54.412 20411-20487/com.namysu.bletest D/mali_winsys: EGLint new_window_surface(egl_winsys_display*, void*, EGLSurface, EGLConfig, egl_winsys_surface**, egl_color_buffer_format*, EGLBoolean) returns 0x3000,  [1440x2560]-format:1
2021-01-25 10:19:54.467 20411-20411/com.namysu.bletest V/InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@8376ac nm : com.namysu.bletest ic=null
2021-01-25 10:19:54.529 20411-20424/com.namysu.bletest D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=4 device=EA:C4:26:58:9A:XX
2021-01-25 10:19:54.563 20411-20424/com.namysu.bletest D/MyBleManager: [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2021-01-25 10:19:54.563 20411-20424/com.namysu.bletest I/MyBleManager: Connected to EA:C4:26:58:9A:10
2021-01-25 10:19:54.565 20411-20424/com.namysu.bletest D/MyBleManager: wait(300)
2021-01-25 10:19:54.918 20411-20411/com.namysu.bletest V/MyBleManager: Discovering services...
2021-01-25 10:19:54.918 20411-20411/com.namysu.bletest D/MyBleManager: gatt.discoverServices()
[... cut ...]
2021-01-25 10:26:56.095 16008-16022/com.namysu.bletest I/MyBleManager: Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 31-39-2E-35-37-23-33-35-39
2021-01-25 10:26:56.096 16008-16008/com.namysu.bletest D/MyBleManager: 19.57#359
2021-01-25 10:27:01.105 16008-16022/com.namysu.bletest I/MyBleManager: Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 31-39-2E-35-37-23-33-35-39
2021-01-25 10:27:01.106 16008-16008/com.namysu.bletest D/MyBleManager: 19.57#359
```

Thank you @zetokz. Could you possibly know why?

Also, after testing on more devices, I will close the issue if there is no problem.

zetokz, 2021-01-25:
@NamSSu I am happy it helped. 

I assume it was because of the PHY request in particular because of its callback. I see that you put 100 seconds as a timeout in connect request and then initiated some additional requests in the `initialize` function. So the library was waiting 100 seconds for initialization to be done and for PHY callback in your case which hasn`t been happening for 100 seconds and the library considered initialization is not done and called disconnect. 

I suppose you can close this issue and start looking at how to deal with the PHY request in your case but that is a whole different story =)

NamSSu, 2021-01-27:
Today, I bought a new device, the Z Fold 2(Android 11.0), and I tested it and it works so well.

For now, I'll buy and test the device as far as I can. if something goes wrong, I'll reopen this issue.

@zetokz Thank you!

closed 2021-01-27 by NamSSu