nordicsemi/Android-BLE-Library #273: Library bug. If onServicesDiscovered gatt callback is called during write characteristic - the result of write operation is missing and write callback is not called.

zetokz, 2021-01-15:
If I call `writeCharacteristic` it will lead in a result to call `BleManagerHandler.nextRequest` where the `WriteRequest` will be stored to private field `private Request request;`. So the library works next way: it stores request and depending of its type - performers different operations in our case WriteRequest ends up with calling `internalWriteCharacteristic` prior saving request. 

As a result, when the `onCharacteristicWrite` gatt callback is called it will check if the saved request is instance of WriteRequest and if so will call its success or fail callback.

But here we have an issue:
If we call write and `gatt.discoverServices()` right away the callback `onServicesDiscovered` is called earlier than `onCharacteristicWrite` and according to its logic 
```java
public final void onServicesDiscovered(@NonNull final BluetoothGatt gatt, final int status) {
  serviceDiscoveryRequested = false;
  ...
  initialize();
  initInProgress = false;
  nextRequest(true);
  ...
}
```
and `nextRequest(true)` according to its logic will rewrite existing request field in our case with null value due to force flag regardless of previous operation in progress.
```java
private synchronized void nextRequest(final boolean force) {
  if (force && operationInProgress) { 
    operationInProgress = awaitingRequest != null;
  }

  if (operationInProgress || initInProgress) {
    return;
  }
  final BluetoothDevice bluetoothDevice = this.bluetoothDevice;

  // Get the first request from the init queue
  Request request = null;
  ...

  boolean result = false;
  operationInProgress = true;
  this.request = request;
  ...
}```

What mean next time when the `onCharacteristicWrite` gatt callback is called right after `onServicesDiscovered` it does not have any effect on the actual write request which callback call we are waiting for as the `request` field is already null.

This situation leads to the case we never receive write callback call and the application hangs on waiting for its callback call.

zetokz, 2021-01-18:
Additional info:
Noticed that this bug reproduces easily when using a splitter and sending a huge amount of data. Like 200000 bytes with mtu 500.

zetokz, 2021-01-18:
Using this gist: https://gist.github.com/zetokz/8be2b4be1444ac186715c4e67802da1a

You can find NordicIssueActivity and related XML file. 
To reproduce the issue run `NordicIssueActivity` and press either SINGLE WRITE button or SEQUNTIAL WRITE button and DISCOVER SERVICES right away after writing. 

**Expected result log must be:
Writing N
Written N**

But the actual result is:
Writing N

That means the final callback is not called and same we see from logs.

philips77, 2021-04-19:
Hello,
Sorry for the super long delay, but I was busy with other tasks.
Why do you call `gatt.discoverServices()` on your own? This method is called internally by the library just after connection.

philips77, 2021-04-20:
OK, as I understand. the `onServiceDiscovered` is called automatically for the second time on some phones, and calling `gatt.discoverServices()` is just a way to force it to happen, is that correct?

philips77, 2021-04-22:
This should be fixed now on *develop* branch. Could you give it a spin?

closed 2021-04-22 by philips77

zetokz, 2021-05-06:
> OK, as I understand. the `onServiceDiscovered` is called automatically for the second time on some phones, and calling `gatt.discoverServices()` is just a way to force it to happen, is that correct?

Yes, that is right.
But my assumption is: even if this function is called manually at any time it should not break the `write request` transition, right?

zetokz, 2021-05-06:
@philips77 I have tested today the issue and it seems to be fixed but some unexpected behavior is still existing and the issue with not called callback can be reproduced. If you want I can share this one with you as well.

philips77, 2021-05-06:
Please, do.