nordicsemi/Android-BLE-Library #276: MyBleManager sample doesn't follow documentation

Maragues, 2021-01-20:
Hi, checking MyBleManager sample, this is the implementation for getGattCallback

```java
    @NonNull
    @Override
    protected BleManagerGattCallback getGattCallback() {
        return new MyManagerGattCallback();
    }
```

But, the documentation of getCallback says the following

> This method must not create a new gatt callback each time it is being invoked, but rather return a single object.

I assume we must return a singleton there, something like 

```java
    private BleManagerGattCallback callback;

    @NonNull
    @Override
    protected BleManagerGattCallback getGattCallback() {
       if(callback == null){
           synchronized(this){
               if(callback == null){
                   callback = new MyManagerGattCallback();
               }
           }
        }

        return callback;
    }
```

Maragues, 2021-01-20:
Perhaps the docs are outdated, since this is now only invoked from the constructor

But, it's protected. So there's the risk of misuse.

philips77, 2021-01-20:
Yes, the doc is outdated. That has change in recent version. If you could create a PR fixing the doc it would be awesome, if not, I'll do that at some point.

Maragues, 2021-01-20:
I created a PR https://github.com/NordicSemiconductor/Android-BLE-Library/pull/277

closed 2021-01-21 by philips77