nordicsemi/Android-BLE-Library #278: BUG. Device disconnected callback is not called if phone loses bonding information

zetokz, 2021-01-21:
Steps to reproduce the issue: 
- bond android with peripheral
- remove bonding on the peripheral only
- connect to the peripheral and try to read secured characteristic

Actual result:
- library prints to the log that the phone lost bonding
- library calls gatt.close without calling disconnect or notifying the client about failed read request
- next connection attempts are not possible because library have `connected = true` flag and each time calling connect it leads to the failed request (see `internalConnect` line 486)

Expected result:
- library notifies about disconnection 
- library allow next connection requests

Logs:
```
2021-01-21 18:47:05.714 D/BluetoothGatt: connect() - device: 80:91:33:63:F1:56, auto: false
2021-01-21 18:47:05.714 D/BluetoothAdapter: isSecureModeEnabled
2021-01-21 18:47:05.715 D/BluetoothGatt: registerApp()
2021-01-21 18:47:05.715 D/BluetoothGatt: registerApp() - UUID=a0cab507-b7d2-4d3d-8de2-ce64e8b2c775
2021-01-21 18:47:05.719 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
2021-01-21 18:47:10.490 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=11 device=80:91:33:63:F1:56
2021-01-21 18:47:10.492 D/MyBLEManager: Nordic|[Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2021-01-21 18:47:10.492 I/MyBLEManager: Nordic|Connected to 80:91:33:63:F1:56
2021-01-21 18:47:10.493 D/MyBLEManager: Nordic|wait(1600)
2021-01-21 18:47:10.967 D/BluetoothGatt: onConnectionUpdated() - Device=80:91:33:63:F1:56 interval=6 latency=0 timeout=500 status=0
2021-01-21 18:47:10.967 I/MyBLEManager: Nordic|Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
2021-01-21 18:47:11.348 D/BluetoothGatt: onConnectionUpdated() - Device=80:91:33:63:F1:56 interval=39 latency=0 timeout=500 status=0
2021-01-21 18:47:11.348 I/MyBLEManager: Nordic|Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)
2021-01-21 18:47:12.097 D/MyBLEManager: Nordic|gatt.discoverServices()
2021-01-21 18:47:12.097 D/BluetoothGatt: discoverServices() - device: 80:91:33:63:F1:56
2021-01-21 18:47:12.105 D/BluetoothGatt: onSearchComplete() = Device=80:91:33:63:F1:56 Status=0
2021-01-21 18:47:12.105 I/MyBLEManager: Nordic|Services discovered
2021-01-21 18:47:12.105 D/MyBLEManager: isServiceSupported: gatt = android.bluetooth.BluetoothGatt@4e62790
2021-01-21 18:47:12.105 D/MyBLEManager: isServiceSupported: service = android.bluetooth.BluetoothGattService@f6ce889
2021-01-21 18:47:12.105 D/MyBLEManager: isServiceSupported: characteristic = android.bluetooth.BluetoothGattCharacteristic@32ef08e
2021-01-21 18:47:12.105 D/MyBLEManager: isServiceSupported: characteristic = android.bluetooth.BluetoothGattCharacteristic@41be0af
2021-01-21 18:47:12.106 D/MyBLEManager: isServiceSupported: characteristic = android.bluetooth.BluetoothGattCharacteristic@4ae2cbc
2021-01-21 18:47:12.106 D/MyBLEManager: isServiceSupported: offering service info
2021-01-21 18:47:12.112 D/MyBLEManager: requireCharacteristic: uuid = e8d06e2a-c987-48f8-93a8-4d18d56b4337, requiredProperty = 2
2021-01-21 18:47:12.113 D/MyBLEManager: handleInitialization: initState = false
2021-01-21 18:47:12.116 D/MyBLEManager: handleInitialization: before reading uuid, bonded devices = [80:91:33:63:F1:56]
2021-01-21 18:47:12.116 D/MyBLEManager: requireCharacteristic: uuid = 8d4675a5-b5fa-42b2-b587-0ee71c46b709, requiredProperty = 2
2021-01-21 18:47:12.116 D/MyBLEManager: Nordic|gatt.readCharacteristic(8d4675a5-b5fa-42b2-b587-0ee71c46b709)
2021-01-21 18:47:12.117 D/MyBLEManager: performConnect: connected
2021-01-21 18:47:12.117 D/MyBLEManager: performConnect: awaitDeviceDisconnected
2021-01-21 18:47:12.323 W/MyBLEManager: Nordic|Authentication required (137)
2021-01-21 18:47:12.323 W/BleManager: Phone has lost bonding information
2021-01-21 18:47:12.348 D/MyBLEManager: Nordic|[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_NONE (10)
2021-01-21 18:47:12.370 D/MyBLEManager: Nordic|gatt.refresh() (hidden)
2021-01-21 18:47:12.371 D/BluetoothGatt: refresh() - device: 80:91:33:63:F1:56
2021-01-21 18:47:12.371 I/MyBLEManager: Nordic|Cache refreshed
2021-01-21 18:47:12.371 D/MyBLEManager: Nordic|gatt.close()
2021-01-21 18:47:12.371 D/BluetoothGatt: close()

2021-01-21 19:06:06.337 E/MyBleManager: performConnect: failure
    Request failed with status -4
    no.nordicsemi.android.ble.exception.RequestFailedException: Request failed with status -4
        at com.myapp.implementation.BleManagerKt$coAwait$$inlined$suspendCancellableCoroutine$lambda$2.onRequestFailed(BleManager.kt:611)
        at no.nordicsemi.android.ble.Request.lambda$notifyFail$2$Request(Request.java:1218)
        at no.nordicsemi.android.ble.-$$Lambda$Request$MU5LE_ePYPVe9rzIdkv_E9MTNdc.run(Unknown Source:6)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:246)
        at android.app.ActivityThread.main(ActivityThread.java:8414)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:596)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
```

philips77, 2021-01-21:
We really appreciate your work on the library. I'm finishing another project now and will handle all your issues when done. Sorry for the delay.

zetokz, 2021-01-22:
> We really appreciate your work on the library. I'm finishing another project now and will handle all your issues when done. Sorry for the delay.

Thanks, I am glad to hear. 
Let me know if I can help you with reproducing issues or need to add logs or anything else.

philips77, 2021-04-16:
Hi,
I finally resumed working on BLE Library. I'm planing on releasing 2.3 next week with all reported issues fixed. I also want to add `-ktx` module with basic support for Kotlin, let's see how will it go.

philips77, 2021-04-22:
It was fixed internally. If possible, could you test it? You can also try after the release and I can still do bug-fix release afterwards :)

closed 2021-04-22 by philips77

mecoFarid, 2022-01-25:
@philips77 Can you please provide a way to handle those exceptions? Those exceptions can't be wrapped with generic `try/catch` because they are thrown from Runnable and can only be caught by `Thread.UnexpectedExceptionHandler`. Also in [Blinky](https://github.com/NordicSemiconductor/Android-nRF-Blinky/), you are not catching any exception. For example, if a user turns off the BL in the middle of the connection it will throw `BluetoothDisabledException` and there is no any place where you have handled this exception in Blinky. What I am missing?