nordicsemi/Android-BLE-Library #279: BUG. Service discovery is called twice even if they discovered successfully (for some phones)

zetokz, 2021-01-22:
Library version: 2.2.4
Android phone: Pixel 2 - Android 11

This pixel 2 phone we have for some reason as you can see from the logs right after connection reports BONDING, BONDED states even though it was bonded before connection. 
This case leads to `discoverServices` call from BleMAnagerHandler.line.348
And later on same call `discoverServices` from BleMAnagerHandler.line.1738

See profs from log =)
That in its turn leads to two calls of `isRequiredServiceSupported` and two calls of `initialize`

```
2021-01-22 16:17:04.860 D/MyBLEManager: Nordic|gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2021-01-22 16:17:04.861 D/BluetoothGatt: connect() - device: 80:91:33:63:F1:56, auto: false
2021-01-22 16:17:04.861 D/BluetoothGatt: registerApp()
2021-01-22 16:17:04.862 D/BluetoothGatt: registerApp() - UUID=
2021-01-22 16:17:04.864 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
2021-01-22 16:17:06.850 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=80:91:33:63:F1:56
2021-01-22 16:17:06.851 D/MyBLEManager: Nordic|[Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2021-01-22 16:17:06.851 I/MyBLEManager: Nordic|Connected to 80:91:33:63:F1:56
2021-01-22 16:17:07.442 D/MyBLEManager: Nordic|[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDING (11)
2021-01-22 16:17:07.450 D/MyBLEManager: Nordic|[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDED (12)
2021-01-22 16:17:07.450 I/MyBLEManager: Nordic|Device bonded
2021-01-22 16:17:09.051 D/MyBLEManager: Nordic|wait(1600)
2021-01-22 16:17:09.052 D/BluetoothGatt: onPhyUpdate() - status=0 address=80:91:33:63:F1:56 txPhy=2 rxPhy=2
2021-01-22 16:17:09.053 I/MyBLEManager: Nordic|PHY updated (TX: LE 2M, RX: LE 2M)
2021-01-22 16:17:17.571 D/MyBLEManager: Nordic|gatt.discoverServices() - called from BleMAnagerHandler.line.348
2021-01-22 16:17:17.571 D/BluetoothGatt: discoverServices() - device: 80:91:33:63:F1:56
2021-01-22 16:17:17.587 D/BluetoothGatt: onSearchComplete() = Device=80:91:33:63:F1:56 Status=0
2021-01-22 16:17:17.588 I/MyBLEManager: Nordic|Services discovered
2021-01-22 16:17:17.588 D/MyBLEManager: isServiceSupported: gatt = android.bluetooth.BluetoothGatt@941cece
2021-01-22 16:17:17.589 D/MyBLEManager: isServiceSupported: service = android.bluetooth.BluetoothGattService@f5c01ef
2021-01-22 16:17:17.590 D/MyBLEManager: isServiceSupported: characteristic = android.bluetooth.BluetoothGattCharacteristic@9dd4fc
2021-01-22 16:17:17.591 D/MyBLEManager: isServiceSupported: characteristic = android.bluetooth.BluetoothGattCharacteristic@100a785
2021-01-22 16:17:17.594 D/MyBLEManager: isServiceSupported: characteristic = android.bluetooth.BluetoothGattCharacteristic@63009da
2021-01-22 16:17:17.594 D/MyBLEManager: isServiceSupported: offering service info
2021-01-22 16:17:58.554 D/MyBLEManager: Nordic|gatt.discoverServices() - called from BleMAnagerHandler.line.1737
2021-01-22 16:17:58.555 D/BluetoothGatt: discoverServices() - device: 80:91:33:63:F1:56
2021-01-22 16:17:58.569 D/BluetoothGatt: onSearchComplete() = Device=80:91:33:63:F1:56 Status=0
2021-01-22 16:17:58.570 I/MyBLEManager: Nordic|Services discovered
2021-01-22 16:17:58.570 D/MyBLEManager: isServiceSupported: gatt = android.bluetooth.BluetoothGatt@941cece
2021-01-22 16:17:58.571 D/MyBLEManager: isServiceSupported: service = android.bluetooth.BluetoothGattService@da93701
2021-01-22 16:17:58.572 D/MyBLEManager: isServiceSupported: characteristic = android.bluetooth.BluetoothGattCharacteristic@b94e9a6
2021-01-22 16:17:58.572 D/MyBLEManager: isServiceSupported: characteristic = android.bluetooth.BluetoothGattCharacteristic@9fc3fe7
2021-01-22 16:17:58.574 D/MyBLEManager: isServiceSupported: characteristic = android.bluetooth.BluetoothGattCharacteristic@f89c794
2021-01-22 16:17:58.574 D/MyBLEManager: isServiceSupported: offering service info
2021-01-22 16:17:58.609 D/MyBLEManager: handleInitialization: initState = false
2021-01-22 16:17:58.611 D/MyBLEManager: performConnect: connected
2021-01-22 16:17:58.611 D/MyBLEManager: performConnect: awaitDeviceDisconnected
2021-01-22 16:17:58.618 D/MyBLEManager: handleInitialization: initState = false
2021-01-22 16:17:58.624 D/MyBLEManager: handleInitialization: before reading uuid, bonded devices = [80:91:33:63:F1:56, 2C:BA:BA:32:FE:59]
2021-01-22 16:17:58.625 D/MyBLEManager: Nordic|gatt.readCharacteristic(uuid)
2021-01-22 16:17:58.630 D/MyBLEManager: handleInitialization: before reading uuid, bonded devices = [80:91:33:63:F1:56, 2C:BA:BA:32:FE:59]
2021-01-22 16:17:58.732 I/MyBLEManager: Nordic|Read Response received from uuid, value: (0x) AF-94-B5-4A-CD-41-49-A9-89-AE-CD-99-2C-13-8C-1D
2021-01-22 16:17:58.732 D/MyBLEManager: readCharacteristic_deprecatedCallback: characteristic = android.bluetooth.BluetoothGattCharacteristic@b94e9a6, data = ���J�AI���͙,�
2021-01-22 16:17:58.733 D/MyBLEManager: Nordic|gatt.readCharacteristic(uuid)
2021-01-22 16:17:58.733 D/MyBLEManager: parseUuid: bytes = [-81, -108, -75, 74, -51, 65, 73, -87, -119, -82, -51, -103, 44, 19, -116, 29]
2021-01-22 16:17:58.737 D/MyBLEManager: handleInitialization: checking is device paired, isPaired = true
2021-01-22 16:17:58.739 D/MyBLEManager: handleInitialization: after reading uuid, bonded devices = [80:91:33:63:F1:56, 2C:BA:BA:32:FE:59]
2021-01-22 16:17:58.739 D/MyBLEManager: handleInitialization: finished, authenticationUuid = uuid
2021-01-22 16:17:58.740 D/BulkDataManager: trackConnectionState: state = com.myble.api.ConnectionState$Connected@43ce04e
2021-01-22 16:17:58.740 D/BleManager: readTransferInfo: start
2021-01-22 16:17:58.859 I/MyBLEManager: Nordic|Read Response received from uuid, value: (0x) AF-94-B5-4A-CD-41-49-A9-89-AE-CD-99-2C-13-8C-1D
2021-01-22 16:17:58.860 D/MyBLEManager: readCharacteristic_deprecatedCallback: characteristic = android.bluetooth.BluetoothGattCharacteristic@b94e9a6, data = ���J�AI���͙,�
2021-01-22 16:17:58.860 D/MyBLEManager: parseUuid: bytes = [-81, -108, -75, 74, -51, 65, 73, -87, -119, -82, -51, -103, 44, 19, -116, 29]
2021-01-22 16:17:58.864 D/MyBLEManager: handleInitialization: checking is device paired, isPaired = true
2021-01-22 16:17:58.865 D/MyBLEManager: handleInitialization: after reading uuid, bonded devices = [80:91:33:63:F1:56, 2C:BA:BA:32:FE:59]
2021-01-22 16:17:58.866 D/MyBLEManager: handleInitialization: finished, authenticationUuid = uuid
```

Techofficer, 2021-03-11:
Hello, I've noticed that discoverServices can be called twice if bonding happens quicker than delay between connection and service discovery.  2 calls to initialisation may lead to problems with reading data and subscription to notifications. 

There are related log lines:
`Connecting...
11:34:49
BrainsculptBleManager
gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
11:34:49
BrainsculptBleManager
onDeviceConnecting: DEVICE_NAME     
11:34:50
BrainsculptBleManager
[Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
11:34:50
BrainsculptBleManager
Connected to F1:06:E9:1A:F9:CE
11:34:50
BrainsculptBleManager
wait(1600)
11:34:50
BrainsculptBleManager
onDeviceConnected: DEVICE_NAME     
11:34:51
BrainsculptBleManager
[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDING (11)
11:34:51
BrainsculptBleManager
onBondingRequired: DEVICE_NAME     
11:34:51
BrainsculptBleManager
[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDED (12)
11:34:51
BrainsculptBleManager
Device bonded
11:34:51
BrainsculptBleManager
onBonded: DEVICE_NAME     
11:34:51
BrainsculptBleManager
Discovering services...
11:34:51
BrainsculptBleManager
gatt.discoverServices()
11:34:51
BrainsculptBleManager
Services discovered
11:34:51
BrainsculptBleManager
Primary service found

....... 

BrainsculptBleManager
Discovering services...
11:34:52
BrainsculptBleManager
gatt.discoverServices()
11:34:52
BrainsculptBleManager
Services discovered
11:34:52
BrainsculptBleManager
Primary service found`

To prevent this you could replace (line 1733 in BleManagerHandler)
`if (connected && gatt.getDevice().getBondState() != BluetoothDevice.BOND_BONDING) {`

with
`if (connected && gatt.getDevice().getBondState() != BluetoothDevice.BOND_BONDING && !serviceDiscoveryRequested) {`

philips77, 2021-04-21:
I applied suggested change and pushed to *develop* branch. This should be released shortly, in April.

philips77, 2021-04-22:
I'll close the issue now, as it has been resolved on *develop* branch. It would be great if you could test it before I'll release 2.3. If not, you I can always release 2.3.1 afterwards if you find this issue again.

closed 2021-04-22 by philips77

zetokz, 2021-04-28:
Thanks for the fixes, I will test them asap
But could you clarify, should I do the fork from develop? As far as I understood you have not created new version of library

philips77, 2021-04-28:
Yes, right now it's only on *develop*. I'm working on -ktx extension. Open for suggestions and PRs!
I will merge it and release a new version in coming days/next week. As it will be passed 1.05, it will not be published on JCenter, only Maven Central. Unless I'm fast :)

zetokz, 2021-05-05:
> Yes, right now it's only on _develop_. I'm working on -ktx extension. Open for suggestions and PRs!
> I will merge it and release a new version in coming days/next week. As it will be passed 1.05, it will not be published on JCenter, only Maven Central. Unless I'm fast :)

Sounds great. I will be testing it today against the sample I prepared spically for testing these issues. 
Great work =)