nordicsemi/Android-BLE-Library #283: NullPointer when init BleManager

kinber95, 2021-02-25:
Hello I add the library in my new Kotlin project, already working with this lib in Java. When I init my BleLegacyManager that extends from nordic BleManager I get following exception ->

```
2021-02-25 13:59:04.261 15123-15123/de.app.haveltec.ilockit E/AndroidRuntime: FATAL EXCEPTION: main
    Process: de.app.haveltec.ilockit, PID: 15123
    java.lang.NullPointerException: Attempt to invoke virtual method 'void no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.init(no.nordicsemi.android.ble.BleManager, android.os.Handler)' on a null object reference
        at no.nordicsemi.android.ble.BleManager.<init>(BleManager.java:168)
        at no.nordicsemi.android.ble.BleManager.<init>(BleManager.java:153)
        at no.nordicsemi.android.ble.livedata.ObservableBleManager.<init>(ObservableBleManager.kt:19)
        at de.app.haveltec.ilockit.ble.BleLegacyManager.<init>(BleLegacyManager.kt:14)
        at de.app.haveltec.ilockit.ui.setup.SetupConnectFragment.onAttach(SetupConnectFragment.kt:102)
```

I init my BleLegacyManager like this -> `val bleLegacyManager = BleLegacyManager(context)`

The BleLegacyManager class -> 

```
class BleLegacyManager(context: Context) : ObservableBleManager(context) {
    val TAG = BleLegacyManager::class.qualifiedName

    ...
    private val mGattCallback = object : BleManagerGattCallback() {
        override fun initialize() {
            super.initialize()
        }

        override fun isRequiredServiceSupported(gatt: BluetoothGatt): Boolean {
            val service = gatt.getService(SERVICE)

            if (service != null) {
              characteristics...
            }

            return service != null
        }

        override fun onDeviceDisconnected() {
            ....
        }
    }

    override fun getGattCallback(): BleManagerGattCallback {
        return mGattCallback
    }
}
```

Did I miss something ? I have the newest library version

philips77, 2021-02-25:
so IMO, your `getGattCallback()` is called before the `mGattCallback` is initiated. Since 2.2 this method is called in the constructor, so the fields of the class are not-yet ready. Instead, define your callback object as a class and return a new instance in `getGattCallback()`.

philips77, 2021-02-25:
Check out https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/MIGRATION.md#changes-in-version-22, point 3.

kinber95, 2021-02-25:
Okay change my code and now it works fine, thanks a lot

```
class BleLegacyManager(context: Context) : BleManager(context) {
    val TAG = BleLegacyManager::class.qualifiedName

    override fun getGattCallback(): BleManagerGattCallback {
        return BleLegacyGattCallback()
    }

    private class BleLegacyGattCallback: BleManagerGattCallback() {
        override fun isRequiredServiceSupported(gatt: BluetoothGatt): Boolean {
            val service = gatt.getService(SERVICE)

            if (service != null) {
              characteristics...
            }
            return service != null
        }

        override fun onDeviceDisconnected() {
        }

    }
}
```

closed 2021-02-26 by kinber95