nordicsemi/Android-BLE-Library #284: ConnectionObserver.onDeviceDisconnected() can never report REASON_TERMINATE_PEER_USER

rickymohk, 2021-03-09:
`ConnectionObserver.onDeviceDisconnected()` always reports `REASON_LINK_LOSS` even if the peripheral initiate disconnection with `BLE_HCI_REMOTE_USER_TERMINATED_CONNECTION`

philips77, 2021-04-22:
`REASON_LINK_LOSS` is reported when you connect with `autoConnect = true` and when the device disconnects without you explicitly calling `disconnect().enqueue()`.
Have a look at:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/1b836ad846ef86fed35fc93807ff8e1056afd57f/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L1518-L1534
The `userDisconnected` flag is set to `true` in here:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/1b836ad846ef86fed35fc93807ff8e1056afd57f/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L587-L588
When `autoConnect` flag is `false`, this is always set to `true`.

rickymohk, 2021-04-22:
The case I am talking about is when the peripheral side initiates disconnection. In that case, `REASON_LINK_LOSS ` is still reported in the central app. There is currently no way to differentiate a connection lost due to bad signal or due to the peripheral explicitly terminating the connection. `REASON_TERMINATE_PEER_USER` instead of `REASON_LINK_LOSS` should be reported when the peripheral explicitly terminates the connection so that the central can determine which case it is.

philips77, 2021-04-22:
What would be expected behavior from the phone side then? Remember, that the connection was initiated with `autoConnect = true`, so the phone will try to reconnect to the device which has disconnected a moment ago.

The reason why you always get link loss in this situation, is that the `autoConnect` flag, in my opinion, should be used for use cases when the connection should be open for as long as possible, and only loosing a signal could in fact cause disconnection. Otherwise, there is no clear behavior what should happen when the remote initiates disconnection: should the library stop reconnecting, should it continue to try to establish connection, or... a user setting?

rickymohk, 2021-04-22:
The behavior could keep the same, but at least give a way for developer to differentiate whether the connection is initiated by the peripheral, in case that information is necessary for the App to determine what to do. In my case, I want to auto reconnect my peripheral if it is disconnected due to let say out of range, but stop auto reconnect if the peripheral explicitly initiates disconnection (e.g. there could be an unpair button or a factory reset button on the peripheral) . Even if the phone keeps auto reconnecting, at least I can write my own code to explicitly call `disconnect().enqueue()` if I know the disconnection was intentionally initiated by the peripheral.

philips77, 2021-04-23:
Let it be.

PR #291 (2021-04-23): Returning proper disconnection reason if peer disconnected
This PR changes the behavior of which status is returned as disconnection reason in case `autoConnect` was set to `true` and the connected peer device initiated disconnection. Before the change `ConnectionObserver.REASON_LINK_LOSS` was returned in any case if the flag was enabled. Now, in case it is the device that caused disconnection, the `ConnectionObserver.REASON_TERMINATE_PEER_USER` will be returned instead.

This PR closes #284.
closed 2021-04-23 by philips77