nordicsemi/Android-BLE-Library #285: NullPointerException on BleManager.onDeviceReady

Maragues, 2021-03-15:
Using latest version

```java.lang.NullPointerException: 
  at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.onDeviceReady (BleManager.java:2215)
  at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.nextRequest (BleManager.java:3158)
  at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.nextRequest (BleManager.java:3222)
  at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.nextRequest (BleManager.java:3222)
  at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.onServicesDiscoveredSafe (BleManager.java:2613)
  at no.nordicsemi.android.ble.MainThreadBluetoothGattCallback.lambda$onServicesDiscovered$1 (MainThreadBluetoothGattCallback.java:112)
  at no.nordicsemi.android.ble.MainThreadBluetoothGattCallback.lambda$onServicesDiscovered$1$MainThreadBluetoothGattCallback (MainThreadBluetoothGattCallback.java)
  at no.nordicsemi.android.ble.-$$Lambda$MainThreadBluetoothGattCallback$BUDsSe1PyUNpgnQBkvDG-U1dE8E.run (-.java:6)
  at android.os.Handler.handleCallback (Handler.java:938)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at android.os.Looper.loop (Looper.java:223)
  at android.app.ActivityThread.main (ActivityThread.java:7660)
  at java.lang.reflect.Method.invoke (Method.java)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:592)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:947)```

Thanks!

philips77, 2021-04-16:
You're not using the latest version of the library. The latest version currently is 2.2.4.

Maragues, 2021-04-16:
Can you tell that from the logs? Because 2.2.4 is the version we use

```
nrfBle             : '2.2.4'

nrfBle                   : "no.nordicsemi.android:ble:$versions.nrfBle",
```

Maragues, 2021-04-16:
Picture from External libraries in Android Studio 
[image]

philips77, 2021-04-16:
In the logs one of the methods is called `onServicesDiscoveredSafe`. This method was renemed to `onServicesDiscovered` in https://github.com/NordicSemiconductor/Android-BLE-Library/commit/f81e1963332173bb3074db5199d51ed9810ea098. It was around 2.2.0-alpha02 when this was changed.

philips77, 2021-04-22:
Hello, Did you figure it out?

Maragues, 2021-04-22:
I no longer remember if this was a local stacktrace or coming from firebase. I guess it's the latter, it was an old app version and I didn't notice

Thanks for your time!

closed 2021-04-22 by Maragues