nordicsemi/Android-BLE-Library #287: MTU not changed after reconnection

jorgeig-space, 2021-03-26:
I have a remote device that initializes with the default MTU. I request a larger MTU in my app at initialization, in my `BleManager` object:

```
protected void initialize() {
            beginAtomicRequestQueue()
                    .add(requestMtu(512)
                            .with((device, mtu) -> log(Log.INFO, "MTU set to " + mtu))
                            .fail((device, status) -> log(Log.WARN, "Requested MTU not supported: " + status)))
                    .done(device -> log(Log.INFO, "Target initialized"))
                    .enqueue();
}
```

This works normally when I do it the first time after opening the app.

If I stop the remote device or go out of range, and then reconnect, the app connects normally and the `initialize()` method is called, but the MTU is not updated. I would expect the MTU to be updated in this case as well.

The log messages show "MTU set to 512", but if I try to write more than the default 23 bytes, I get:
````
E/BleManager: onCharacteristicWrite error 6
E/MyBleManager: Error (0x6): GATT REQ NOT SUPPORTED
````

If I add two requests to force the MTU change, then everything works correctly, but only if both requests are sent. Sending only the 512 bytes request has no effect.
```
requestMtu(28).enqueue();
requestMtu(512).enqueue();
```

I am not sure if this is a library issue or an Android issue. I do not think it is a remote device issue since there is no outgoing BLE traffic after the re-connection if I do not do the workaround.

ding209, 2021-04-12:
I ran into the same problem,I temporarily use “overrideMtu(23)” to solve it

jeffbk, 2021-04-14:
I ran into this too recently.  I think this is related to issue #249.

philips77, 2021-04-16:
Hi,
This issue was fixed in *develop* branch, but for some reason that was not released yet.
I just resumed working on this project, so I'll go through all issues, fix bugs and will do a release.

philips77, 2021-04-16:
Have a look at: https://github.com/NordicSemiconductor/Android-BLE-Library/commit/f89674cb4af3321df46dc6a3d447e2aced2af78c

philips77, 2021-04-16:
I'll close it as it as solved on *develop*.

closed 2021-04-16 by philips77