nordicsemi/Android-BLE-Library #288: notification callback never get called.

IHNEL, 2021-04-16:
I implement my BleManager at client side to connect to another android device acts as Ble server. The client device can read or write data but never get called when server notify new changes. If I use built-in Gatt callback, method `onCharacteristicChanged` get called as expected. However, with this library, `setNotificationCallback` never get called. Here is my code where the characteristic is setup:

```

            // Initialize your device here. Often you need to enable notifications and set required
            // MTU or write some initial data. Do it here.
            override fun initialize() {
                // You may enqueue multiple operations. A queue ensures that all operations are
                // performed one after another, but it is not required.
                beginAtomicRequestQueue()
                    .add(
                        requestMtu(247) // Remember, GATT needs 3 bytes extra. This will allow packet size of 244 bytes.
                            .with { device: BluetoothDevice?, mtu: Int ->
                                log(Log.INFO, "MTU set to $mtu")
                            }
                            .fail { device: BluetoothDevice?, status: Int ->
                                log(Log.WARN, "Requested MTU not supported: $status")
                            }
                    )
                    .add(enableNotifications(settingChar))
                    .add(enableNotifications(playingChar))
                    .add(enableNotifications(alarmChar))
                    .done(SuccessCallback { device: BluetoothDevice? ->
                        log(Log.INFO, "Target initialized")
                    })
                    .fail { device, status ->
                        log(Log.WARN, "Fail with status $status")
                    }
                    .enqueue()

                setNotificationCallback(settingChar).with { device, data ->
                    onNewAppSettingsData(data)
                }
                setNotificationCallback(playingChar).with { device, data ->
                    Log.d("TEST", "On Notification")
                    onNewPlayingStatus(data)
                }
                setNotificationCallback(alarmChar).with { device, data ->
                    onNewAlarmStatus(data)
                }
                requestAppSettings()
                turnOnAllNotification()
            }

    private fun turnOnAllNotification() {
        settingChar?.getDescriptor(DESCRIPTOR_UUID)?.let {
            writeDescriptor(it, BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE).enqueue()
        }
        playingChar?.getDescriptor(DESCRIPTOR_UUID)?.let {
            writeDescriptor(it, BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE).enqueue()
        }
        alarmChar?.getDescriptor(DESCRIPTOR_UUID)?.let {
            writeDescriptor(it, BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE).enqueue()
        }
    }
```

a side note, at server side, this callback function get called when it send new notification data (which is pretty fine):

```
override fun onNotificationSent(device: BluetoothDevice?, status: Int) {
            super.onNotificationSent(device, status)
            Log.d("TEST","On notification sent to device ${device?.name} $status")
        }

```

Do I setup my characteristic wrong?

IHNEL, 2021-04-16:
I did this workaround after a headache day: I call function `gatt.setCharacteristicNotification` directly instead of `enableNotifications` inside `isSupportNotification` callback function.  I think there is a bug inside this library where it fails to turn on notification itself. I'm using latest versiopn 2.2.4, by the way.

`gatt.setCharacteristicNotification(playingChar,true)`

```

            private fun isSupportNotification(chrs: BluetoothGattCharacteristic?): Boolean {
                if (chrs == null) return false
                val canNotify = chrs.properties and BluetoothGattCharacteristic.PROPERTY_NOTIFY != 0
                if (!canNotify)
                    log(Log.WARN, "Characteristic ${chrs.uuid} doesn't support notification")
                return canNotify
            }

            // This method will be called when the device is connected and services are discovered.
            // You need to obtain references to the characteristics and descriptors that you will use.
            // Return true if all required services are found, false otherwise.
            public override fun isRequiredServiceSupported(gatt: BluetoothGatt): Boolean {
                val service = gatt.getService(SERVICE_UUID)
                var notify = false
                if (service != null) {
                    settingChar = service.getCharacteristic(SETTINGS_CHAR)
                    playingChar = service.getCharacteristic(PLAYING_CHAR)
                    alarmChar = service.getCharacteristic(ALARM_CHAR)
                }
                gatt.setCharacteristicNotification(settingChar,true)
                gatt.setCharacteristicNotification(playingChar,true)
                gatt.setCharacteristicNotification(alarmChar,true)
                return settingChar != null
                        && playingChar != null
                        && alarmChar != null
                        && isSupportNotification(settingChar)
                        && isSupportNotification(playingChar)
                        && isSupportNotification(alarmChar)
            }
```

jorgeig-space, 2021-04-16:
You are missing re-enabling the notifications. The documentation for `BleManager.setNotificationCallback` states:

> After assigning the notifications callback, notifications must be enabled using enableNotifications(BluetoothGattCharacteristic). This applies also when they were already enabled on the remote side.

philips77, 2021-04-16:
First of all, you should not need:
```kotlin

    private fun turnOnAllNotification() {
        settingChar?.getDescriptor(DESCRIPTOR_UUID)?.let {
            writeDescriptor(it, BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE).enqueue()
        }
        playingChar?.getDescriptor(DESCRIPTOR_UUID)?.let {
            writeDescriptor(it, BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE).enqueue()
        }
        alarmChar?.getDescriptor(DESCRIPTOR_UUID)?.let {
            writeDescriptor(it, BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE).enqueue()
        }
    }
```
This is done automatically when you add `enableNotificatons` in `initialize()`.

You should also not need calling `gatt.setCharacteristicNotification(XXX, true)`, as they are also called automatically.

philips77, 2021-04-16:
Do you have Client Characteristic Configuration Descriptors (CCCD) in all those characteristics? Are you sure none of them is `null`?

philips77, 2021-04-16:
You may also add `.fail` callback to each `enableNotifications` request to check which one is failing.

IHNEL, 2021-04-16:
hi @philips77. Thanks for your answers
If I disable function `turnOnAllNotification` and `gatt.setCharacteristicNotification(XXX, true)`, notification doesn't work. It fails for all characteristics.

I didn't add `CCCD` for my custom characteristics at server side.  What I did is adding a custom descriptor into characteristics and listen to the write-event to descriptor and add the client device to a list. When user changes data at server side, the app will iterate all registered devices and send notification. Be aware that I don't use this library at server side (actually I tried but it's hard for me to following and setup).

Here is how I did at server side:

```
val Settings_Service: UUID = UUID.fromString( "00000000-1000-2000-3000-111122223333")
    /**
     * Mandatory Client Characteristic Config Descriptor.
     * Client needs to write data to this characteristic to sign-up for value changed notifications
     */
    val CLIENT_CONFIG: UUID = UUID.fromString("00010000-1000-2000-3000-111122223333")
    val APP_SETTINGS = UUID.fromString("00000004-1000-2000-3000-111122223333")
 private val properties = BluetoothGattCharacteristic.PROPERTY_READ or BluetoothGattCharacteristic.PROPERTY_WRITE or BluetoothGattCharacteristic.PROPERTY_NOTIFY
    private val permissions = BluetoothGattCharacteristic.PERMISSION_READ or BluetoothGattCharacteristic.PERMISSION_WRITE
    val service = BluetoothGattService(
            Settings_Service,
            BluetoothGattService.SERVICE_TYPE_PRIMARY
        )
    val characteristic = BluetoothGattCharacteristic(APP_SETTINGS, properties, permissions)
            val configDescriptor = BluetoothGattDescriptor(CLIENT_CONFIG, permissions)
            characteristic.addDescriptor(configDescriptor)
            service.addCharacteristic(characteristic)
      //The same for other characteristics

```
     [...]

     private val gattServerCallback = object : BluetoothGattServerCallback() {
             override fun onDescriptorWriteRequest(
            device: BluetoothDevice, requestId: Int,
            descriptor: BluetoothGattDescriptor,
            preparedWrite: Boolean, responseNeeded: Boolean,
            offset: Int, value: ByteArray
        ) {
            fun sendRespond(status: Int, data: ByteArray?) {
                bluetoothGattServer?.sendResponse(device, requestId, status, 0, data)
            }
           
            if (GattProfile.CLIENT_CONFIG == descriptor.uuid) {
                if (ENABLE_NOTIFICATION_VALUE.contentEquals(value)) {
                    registeredDevices.add(device)
                } else if (DISABLE_NOTIFICATION_VALUE.contentEquals(value)) {
                    registeredDevices.remove(device)
                }

                if (responseNeeded) {
                    sendRespond(GATT_SUCCESS, value)
                }
            } 
        }

     }

IHNEL, 2021-04-16:
FYI, I followed this sample to implement my custom Bluetooth LE server on android device: https://github.com/androidthings/sample-bluetooth-le-gattserver (It's android smart watch).

philips77, 2021-04-19:
Hi,
For the `enableNotifications` to work you need to have the CCCD descriptor UUID under each characteristic with NOTIFY/INDICATE property.
In this document: https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf you'll find, that the 16-bit UUID for this descriptor must be 0x2902. Search for Client Characteristic Configuration descriptor. Also, in the code you're referring to the UUID is correct: https://github.com/androidthings/sample-bluetooth-le-gattserver/blob/b37571a2e28b538371f1f87a24e1eb202b9ddd0d/kotlin/app/src/main/kotlin/com/example/androidthings/gattserver/TimeProfile.kt#L39.

Using a custom UUID for this descriptor is not really useful and is problematic, as standard methods won't work, as you see.

IHNEL, 2021-04-19:
Thank you very much @philips77 
Lets me change the descriptor uuid and try again. Hopefully it rescues me from the nightmare 😅

closed 2021-04-21 by philips77

jeremycartergf, 2025-02-04:
A side note for anyone else, be sure to call `super.initialize()` before setting up your notifications.