nordicsemi/Android-BLE-Library #290: Is there any full example which use this library at both client side and server side?

IHNEL, 2021-04-19:
Hi. I am really like this lib because of good callback structure and job queue. For now I use it at client device while i use built-in gatt server callback at server (another android device). 
I tried to setup the server side with this library but guidelines in Readme isn’t clear enough to me (it’s my bad). Is there any full sample code which I can dive in and do the same thing?

philips77, 2021-04-20:
Have a look at samples in *develop* branch: https://github.com/NordicSemiconductor/Android-BLE-Library/tree/develop/examples (a specially the second one). I'll gladly help you if you need help. I tried my best explaining it in Readme, sorry it's not enough.

Btw, the lib was design to work in client only and then I added client+server mode. Only server is still not supported. I'm thinking of rewriting the whole lib to kotlin as version 3 with severe API changes. Let's see how this goes.

IHNEL, 2021-04-20:
> Btw, the lib was design to work in client only and then I added client+server mode. Only server is still not supported. I'm thinking of rewriting the whole lib to kotlin as version 3 with severe API changes. Let's see how this goes.

Thank you @philips77 , what do you mean about `client+server` mode? is it possible to use the lib with server mode in 1 device, and then with client mode on other device?

philips77, 2021-04-20:
By *client* mode, I mean when you connect to a device which has some services. The easiest example.
*Server* mode is when you have some GATT services on your Android phone, you advertise, and some other device connects, does service discovery and starts using your services. This is not yet supported in the BLE library, as it was never intended to do, so there's a lot of things that would have to be changed in the API. I'll think about this mode when designing version 3.
*Client+server* mode is when you want to use "data" on remote device, but also have some "data" accessible for the other side. In that case, before doint anything else, you [set up a GATT server](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/v2.2.4/README.md#adding-gatt-server-support).  Then:
1. If you are connecting to a remote device, you use `setServerManager(...)` and connect as in *client* mode.
2. If you advertised, you'll get `onDeviceConnectedToServer` callback, from where you may connect as in 1.

In *client+server* mode you still connect to the remote device, but it has a chance to write/read data from the phone.

You may also mix requests, e.g.:
```kotlin
  writeCharacteristic(...)  // a client method: client writes data to server
     .trigger(waitForWrite(...)) // a server method: your local server waits until the remote side will write something to your server characteristic.
     .enqueue()
```

philips77, 2021-04-20:
The *server*-only mode is not supported, as in the current version you set the `BluetoothDevice` in `connect` method. Also, I personally don't like APIs where you can do something illegal, like call `writeCharacteristic` when you're not connected as client. But on the other hand, you can do this now with calling server-method when only using client... Have a look at #231. I'll need to think about the new design.

philips77, 2021-04-20:
Also, if you want to t use *server*-only, you may still use *client+server* and just... not use client. So your `initialize()` would do nothing, your `isRequiredServiceSupported` would return always `true`, etc.

closed 2021-04-22 by philips77