nordicsemi/Android-BLE-Library #294: Is there a demo for connecting and sending messages?

xhtlxa, 2021-04-28:
Is there a demo for connecting and sending messages?

philips77, 2021-04-28:
Check out samples on *develop* branch: https://github.com/NordicSemiconductor/Android-BLE-Library/tree/develop/examples
Also check out nRF Blinky: https://github.com/NordicSemiconductor/Android-nRF-Blinky/tree/develop

philips77, 2021-04-28:
I'll release a new version (one that's now on *develop* branch + some more features/bug fixes) in next few days.

closed 2021-04-28 by philips77

WangShiJieAOTM, 2022-12-04:
> 

page not found , have new link?

WangShiJieAOTM, 2022-12-07:
The example folder contains the kotlin version. Is there a Java version?

philips77, 2022-12-08:
You may find Java version of nRF Blinky at tag *2.7.0* here: https://github.com/NordicSemiconductor/Android-nRF-Blinky/tree/2.7.0

Here's the `BlinkyManager` in Java: https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/2.7.0/app/src/main/java/no/nordicsemi/android/blinky/profile/BlinkyManager.java