nordicsemi/Android-BLE-Library #295: Proposal for ktx extensions for the library in connection with coroutines

zetokz, 2021-05-05:
```
suspend fun Request.coAwait(): BluetoothDevice = suspendCancellableCoroutine { continuation ->
    done { continuation.resume(it) }
        .invalid { continuation.resumeWithException(InvalidRequestException(this)) }
        .fail { _, status ->
            val exception = when (status) {
                FailCallback.REASON_BLUETOOTH_DISABLED -> BluetoothDisabledException()
                FailCallback.REASON_DEVICE_DISCONNECTED -> DeviceDisconnectedException()
                else -> RequestFailedException(this, status)
            }
            continuation.resumeWithException(exception)
        }
        .enqueue()
}

suspend fun ReadRequest.coAwaitWithResult(): ByteArray? {
    var result: ByteArray? = null
    with { _, data -> result = data.value }.coAwait()
    return result
}

suspend fun ReadRequest.coAwaitWithResult(): ByteArray? {
    var result: ByteArray? = null
    with { _, data -> result = data.value }.coAwait()
    return result
}
``` 

And similar for other types of requests :)

philips77, 2021-05-06:
Much love!

philips77, 2021-05-11:
Please, have a look and review.

closed 2021-07-21 by philips77

mecoFarid, 2022-01-25:
Do you guys realize that those exceptions can't be caught?! Well, unless you use Thread.UncaughExceptionHandler.