nordicsemi/Android-BLE-Library #298: NullPointerException

979451341, 2021-05-21:
java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.bluetooth.BluetoothGatt.discoverServices()' on a null object reference
        at no.nordicsemi.android.ble.BleManagerHandler.lambda$nextRequest$27$BleManagerHandler(BleManagerHandler.java:3194)
        at no.nordicsemi.android.ble.-$$Lambda$BleManagerHandler$Y52Tq_9iTPwnU65wNXgeetbigjU.run(Unknown Source:6)
        at android.os.Handler.handleCallback(Handler.java:900)
        at android.os.Handler.dispatchMessage(Handler.java:103)
        at android.os.Looper.loop(Looper.java:219)
        at android.app.ActivityThread.main(ActivityThread.java:8393)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1055)

philips77, 2021-08-31:
We will look into it for the next release.

philips77, 2021-08-31:
Based on the line numbers, you're not using the latest version of the library. Could you try with 2.3.0?

closed 2022-02-02 by philips77