nordicsemi/Android-BLE-Library #309: Kotlin cannot access package-private

ddB0515, 2021-07-29:
Hi

As I have started using library having issue with access to ObservableBleManager and getting this error, not sure what is causing it...
Anyone had same issue and how to resolve it???

Cannot access 'ObservableBleManager': it is package-private in 'no.nordicsemi.android.ble.livedata'
[image]

Using Kotlin 1.5.21 (if is needed) and using via maven (not as module)

philips77, 2021-07-30:
Yes, we found it as well. I'll fix it when I can. Sorry for that. For now you may copy the class to your project.

philips77, 2021-09-23:
This has now been fixed and will be released as 2.3.1.

PR #321 (2021-09-23): Making ObservableBleManager public
This PR fixes #309.
PR #325 (2021-09-23): Version 2.3.1
This version introduces the following changes:
* #321 which fixes #309
* #322 which fixes #316
* #323 
* #324 which fixes #317
closed 2021-09-23 by philips77