nordicsemi/Android-BLE-Library #311: Separate Location and Bluetooth permission with Android 12

SaqibJDev, 2021-08-02:
Hi,
Since Android 12, location permission has been separated from Bluetooth permission. It would be great if the BLE library also reflects this change to enable apps to work without location permission when required.

philips77, 2021-08-31:
Managing permissions is the responsibility of the applications. Currently, the app actually requests BLUETOOTH and BLUETOOTH_ADMIN permissions, but you're right, that could be removed leaving the app of charge of everything.
Location permission is only required for scanning, and BLE Library does not scan at all. On Android 12 it's the BLUETOOTH_CONNECT permission that needs to be requested now in runtime.

philips77, 2021-08-31:
Task: Remove BLUETOOTH and BLUETOOTH_ADMIN permissions from AndroidManifest in the library. Look into new permissions and what exceptions are thrown in case permission is not granted.

philips77, 2021-09-23:
Commit 4b01f25469573f74762bed40c0a15ea5a958cbef has added info about required permissions and set `maxSdkVersion` to 30 for `android.permission.BLUETOOTH` permission.

closed 2021-09-23 by philips77