nordicsemi/Android-BLE-Library #312: [Question] Is it possible to wait for a specific response packet

Chevalier-Maxime, 2021-08-04:
Hi there,
Thank you for this library which is really nice to use ! 

I have a question about synchronous BLE operation. I'd like to perform a BLE write and then wait for a specific response on a characteristic (I can have multiple responses on this characteristic). Is there a way to "filter" out packet that I don't want to deals with a this moment (they are also received in as notifications). 

I saw a "filter" method but I'd like to have the full packet (my BLE packets needs to be merged, but filter does not wait for the full packet).

Is there a way to achieve this with the library ? 

Thank you, have a nice day.

philips77, 2021-08-04:
I suppose no... Actually, you're right. There should be two "filters". For each packet before merging and for merged packets. I was trying to come up with a solution using the current API, e.g. using a custom merger, but I don't see it working.
I think the only thing you may do is just ignore merged packets on your side, as this is what that feature would do internally.

philips77, 2021-08-04:
Let's write a reminder for myself: implement a way to filter merged packets. It can be done by ignoring unwanted on the user's side, but the lib should allow for that as well.

Chevalier-Maxime, 2021-08-05:
Thanks for the feedback ! 
I'm not sure to understand what you means by 
> "I think the only thing you may do is just ignore merged packets on your side"

Once I receive my merged packet the WaitForValueChangeRequest is unblocked, no ?

I currently do something like this (might not be the expected way) :
```
waitForValueChangedRequest = waitForIndication(responseCharacteristic);
waitForValueChangedRequest.merge(dataMerger);
writeRequest = writeCharacteristic(requestCharacteristic, requestData);
if (dataSplitter != null) {
      writeRequest.split(dataSplitter::chunk);
}
// Call request
final ReadResponse readResponse = waitForValueChangedRequest.trigger(writeRequest).timeout(timeout).await(new ReadResponse());
```

philips77, 2021-08-31:
> Once I receive my merged packet the WaitForValueChangeRequest is unblocked, no ? I can't block it again ?

True, if you're using `waitForIndication`, not `setIndicationCallback` than yes... Something to address in the next release.

philips77, 2022-03-21:
I've stated working on this feature.

closed 2022-03-21 by philips77