nordicsemi/Android-BLE-Library #313: How to run a block of code when a characteristic is written

jppdpf, 2021-08-05:
Hi, 

I'm new using this library and Kotlin so I need a little help trying to understand one thing.

I used the GATT Server Example as base. I added have four characteristics - because I can't save 2048 bytes in a single one - and I want to attach a block of code every time a characteristic is written.

How/Where can I implement that?

Thank you very much for your time.

philips77, 2021-08-09:
So this library is designed to work either as a client only, where a device has services and the phone discovers them and use them, or as client + server mode, where both sides may have services and the phone is using services exposed by the device and is also exposing some services for the device. The server only mode is only possible as client + server where the phone ignores the device's services. That is what the GATT Server Example is doing.

You may set up the server by adding any number of services and characteristics (by any I mean less than few dozens). Than connect to the device, or start advertising on the phone, so the device connects to the Android. In the latter case you'll get a [callback](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/58dde71866bc0e796c396d99aa3cceac5e940f44/ble/src/main/java/no/nordicsemi/android/ble/observer/ServerObserver.java#L21) in the server delegate, from where you should [connect to the device](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/58dde71866bc0e796c396d99aa3cceac5e940f44/examples/ble-gatt-server/src/main/java/no/nordicsemi/android/ble/ble_gatt_server/GattService.kt#L235) (this is the client connection, which doesn't have to be used, but it's required for the Ble Manager to work properly. When connected (either way), you can use [setWriteCallback(...)](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/58dde71866bc0e796c396d99aa3cceac5e940f44/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L725) to get callbacks whenever a server characteristic is written.

philips77, 2021-08-09:
Also, you may have just one characteristic and write it multiple times. You can use merger to merge them into a single value, if needed. I don't know what type of values are you writing. 
To write 2048 bytes you may either use few long writes to a characteristic with WRITE (with response) property, or a lot of writes to a characteristic with WRITE WITHOUT RESPONSE property, each packet of size MTU-3 bytes.

jppdpf, 2021-08-12:
Thank you very much for the time and explanation.
I am still trying to make this work. 
Do you have an example of this by any chance (using merger, or setting a write callback)?

philips77, 2021-08-13:
Just call
```kotlin
setWriteCallback(characteristic)
   .merge(YourMerger())
   .with(callbackWhichWillGetMergedMessage)
```
Check [DataMerger](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/58dde71866bc0e796c396d99aa3cceac5e940f44/ble/src/main/java/no/nordicsemi/android/ble/data/DataMerger.java) interface or [an example of one](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/58dde71866bc0e796c396d99aa3cceac5e940f44/ble/src/main/java/no/nordicsemi/android/ble/data/JsonMerger.java).

jppdpf, 2021-08-13:
I find a solution :)
Not one simple like yours, a more java one.
Using the base of the ble-gatt-server-example the following is what I did.
Inside ServerConnection.
[image]

[image]

And inside of ServerManager
[image]

A very big thanks for the help! For now i will stick with my solution just because it works, later on when i implement a serious solution I will try yours for sure!!

closed 2021-08-13 by philips77

georrge1994, 2021-11-27:
@jppdpf 
[image]