nordicsemi/Android-BLE-Library #314: Discovering new services on an active ble device connection

Cplantijn, 2021-08-16:
I am trying to find a way to "refresh" services on an existing, bonded connection. I am interfacing with an Ble mesh device that emits new services after it's identified and provisioned. I would like to avoid disconnecting from the device and reconnecting to discover new services UUIDs. Is there an API in the library for me to get new service UUIDS? Similar to calling
`BluetoothGatt.discoverServices()` and waiting for `BluetoothGatt.onServicesDiscovered` Thanks.

philips77, 2021-08-17:
There is https://github.com/NordicSemiconductor/Android-BLE-Library/blob/58dde71866bc0e796c396d99aa3cceac5e940f44/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L1841
On Android 12 you shouldn't need to call that, as there is `onServiceChanged` callback in `BluetoothGattCallback` which this lib makes use of. That was, however, never tested.

The bonded device should send Service Changed indication to inform client about the change.

Cplantijn, 2021-08-17:
That worked well for Android 11 for me, @philips77. Thank you

closed 2021-08-17 by Cplantijn