nordicsemi/Android-BLE-Library #316: Can't Disable Indication

GsMoMo, 2021-08-23:
When we try to disable the indication of a characteristic by calling the function "disableIndications" in "BleManagerHandler.java", the function always returns the error code -3 (The null attribute). Therefore, I traced back to the code base and found the function "disableIndications" is as same as "disableNotification". From my point of view, when it tries to get the descriptor it should use the same property type as the function "enableIndications" does as below:

`final BluetoothGattDescriptor descriptor = getCccd(characteristic, BluetoothGattCharacteristic.PROPERTY_INDICATE);`

Could you check this issue? Thanks.

philips77, 2021-08-23:
You're right. Looks like a bug here:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/58dde71866bc0e796c396d99aa3cceac5e940f44/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L811
This is called from:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/58dde71866bc0e796c396d99aa3cceac5e940f44/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L844-L847

philips77, 2021-08-23:
Thank you for the report! I'll fix it asap.

GsMoMo, 2021-08-24:
Thanks a lot!

philips77, 2021-09-23:
Will be released in 2.3.1.

PR #322 (2021-09-23): Disabling indications fixed
This PR fixes #316.
PR #325 (2021-09-23): Version 2.3.1
This version introduces the following changes:
* #321 which fixes #309
* #322 which fixes #316
* #323 
* #324 which fixes #317
closed 2021-09-23 by philips77