nordicsemi/Android-BLE-Library #317: Can't specify write type in BleManager.writeCharacteristic

olivier-basille, 2021-08-30:
Hi there, thanks for the great BLE library!

I'm trying to write to a characteristic without response, and I can see that some of WriteRequest's constructors have a "writeType" parameter that should let me pass `BluetoothGattCharacteristic.WRITE_TYPE_NO_RESPONSE`.

But I don't find any version of `BleManager.writeCharacteristic()` that take the same parameter and let me specify to write without response.

Is there something I'm missing or workaround for this?
Thanks!

philips77, 2021-08-30:
Ok, you got me. I went through the code quickly and couldn't find it, but it used to be there. I can see that the `WriteRequest` class takes write type as a parameter in one of the constructors. That one should be used in the API.

philips77, 2021-08-30:
Ok, from here
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/1c8339013e678a864302209618435de5707207dd/ble/src/main/java/no/nordicsemi/android/ble/Request.java#L330
it looks like the write type is taken from the characteristic. So you need to call 
```java
characteristic.setWriteType(...);
writeCharacteristic(characteristic, data);
```

olivier-basille, 2021-08-30:
Thank you @philips77 for the quick and useful response!

It's worth noting that this behavior can create a race condition as opposed to passing the write type as an argument to `writeCharacteristic()`.  (For example when there are 2 threads writing to the same characteristic, one with response, and the other without response)

philips77, 2021-08-31:
Yes, I'm aware of it. I'll fix it in the next release. There used to be a version of this method with `writeType` 
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/828fc9a0ec6e69c6dd7082b2107b3680da85a601/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L719
and it vanished later...

philips77, 2021-08-31:
I'll keep it open until I fix it. Once again, thanks for reporting!

philips77, 2021-09-23:
Will be released in 2.3.1.

PR #324 (2021-09-23): Adding write type param to writeCharacteristic methods
This PR fixes #317. The `writeType` param is now passed directly to the `writeCharacteristic` methods.
Existing `writeCharacteristic` methods without `writeType` have been deprecated.
PR #325 (2021-09-23): Version 2.3.1
This version introduces the following changes:
* #321 which fixes #309
* #322 which fixes #316
* #323 
* #324 which fixes #317
closed 2021-09-23 by philips77