nordicsemi/Android-BLE-Library #318: Unable to disconnect.

ELIX1337, 2021-09-08:
The application is written following the example of nrfBlinky.
Logs says the onDeviceDisconnected() method of ObservableBleManager works fine and disconnect occurs, I also get message:
E/bt_stack: [ERROR:gatt_api.cc(1268)] GATT_Disconnect conn_id=0x000d
But in fact the device is not visible for scanning and becomes visible after the application is closed and I get these messages:
E/bt_stack: [ERROR:gatt_api.cc(1268)] GATT_Disconnect conn_id=0x0010
E/bt_stack: [ERROR:gatt_api.cc(1268)] GATT_Disconnect conn_id=0x000e
E/bt_stack: [ERROR:gatt_api.cc(1268)] GATT_Disconnect conn_id=0x000f
E/bt_stack: [ERROR:gatt_api.cc(1268)] GATT_Disconnect conn_id=0x0011

I know that the device can keep connections for other purposes of the smartphone and a disconnect request from one app does not guarantee physical disconnection. But this is a custom device with custom firmware and this behavior (as far as I understand) should not be

Galaxy M12
Android 11

philips77, 2021-09-08:
It may be that you have a leak and connect to the device multiple times (have multiple instances of your `BleManager`). The physical connection will close when the last client has disconnected, and seems like you have 5 of them.
Also, as you wrote, it may be another app that also connected to your device. nRF Connect can do this, but since Android 8.0 it uses opportunistic connection, so should not take over the connection, but perhaps other apps?

To make sure the device in fact disconnects, you may implement a graceful disconnection from the device side. Your app, instead of calling `disconnect()` would write some Op Code after which the device would terminate the connection. This is the only way an app can guarantee that the connection has closed.

ELIX1337, 2021-09-08:
@philips77 

Yes, i just found out that the reason is most likely a leak. I will try to solve it, thank you

closed 2021-09-23 by philips77