nordicsemi/Android-BLE-Library #330: BLE Scan Not working on android 12

kirankirvelrm, 2021-11-22:
I use this library in my application.
Now I am testing the application with android 12 and in scan only it's not working. Not showing a single device in a scan. I use UDID based scan and connectivity.

Yazazzello, 2022-02-01:
If check the error code from library - you'll see that it's related to permissions. On Android 12 Bluetooth permissions have been changed. check the documentation - https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

philips77, 2022-02-01:
Hello,
Thank you for answering this old issue, I must have missed it.
This library has nothing to do with scanning for BLE Devices, but @Yazazzello is right. Check your permissions and location service. If you need to scan for iBeacons or Eddystone beacons, make sure you request for both `BLUETOOTH_SCAN` and `ACCESS_FINE_LOCATION` permissions. Otherwise, use the `neverForLocation` flag in the Android Manifest.

closed 2022-02-01 by philips77