nordicsemi/Android-BLE-Library #333: Bluetooth callbacks are delayed

jswn0203, 2021-11-30:
When the device I connected was turned off, there was a delay in the code callback of bluetooth status, about four or five seconds. I don't know whether it was the device or the code of this library. I hope you can help me deal with and analyze the cause

philips77, 2021-11-30:
This is called a supervision timeout. It is set int the Connection Parameter Update requests, negotiated between the central and peripheral. 4 seconds is standard time, which can be lowered from the peripheral side if necessary.
Android lets you control the connection parameters only in limited scope, using:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/122319c029746f86e210874da642d5477b003817/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L1819-L1849
The supervision timeout used to be 20 sec, like in the above documentation, but on Android 12 I can see it's set to 5 seconds. I don't know when has it been lowered.

philips77, 2022-02-02:
Updated supervision timeout comments: https://github.com/NordicSemiconductor/Android-BLE-Library/commit/7e3b4f5c5505f151f358989034328159e5a9b62f

closed 2022-02-02 by philips77