nordicsemi/Android-BLE-Library #334: Get access to cancelConnection

georrge1994, 2021-12-01:
Hi, team. Thank you for your library - it's really helpful stuff. Today I faced with problem: I'm need option to `cancelConnection` for already connected device. Gatt server could be get from bleManager, but `getServer` method has a package visibility. I don't want to import full lib, so I just get access via reflection. Maybe in next versions you will add cancel-method or make server wirh protected visibility. Thx.

georrge1994, 2021-12-01:
And seems, that `cancelConnection ` doesn't work for android gatt server. Crap https://issuetracker.google.com/issues/37127644?pli=1

philips77, 2021-12-01:
If the device is already connected, you need to call `disconnect().enqueue()`. The `cancelConnection()` is only for a pending connection attempt, as a client.
As this library supports actually only Client-only and Client+Server modes, not Server-only, disconnecting the client will also disconnect the server.

But what exactly do you want to achieve? Could you elaborate on that a bit more, perhaps this can be done some different way. Please help me understand the issue.

philips77, 2021-12-01:
As I understand, you'd like to get access to [`BluetoothGattServer.cancelConnection(BluetoothDevice)`](https://developer.android.com/reference/android/bluetooth/BluetoothGattServer#cancelConnection(android.bluetooth.BluetoothDevice)) method. Let me write you how this method actually works, as this is completely not clear from the documentation.

When a device is connected to the phone, the Android system is maintaining the physical link to the device. Apps can get a "virtual" connection (`BluetoothGatt` object for client connections, or a connection to the GATT server), which may be **strong** or **weak**.
If you connect to a device as a client, using `connectGatt`, you're effectively making a strong connection. Android will keep the physical connection until the last strong connection exists. If you'd check sources of `BluetoothDevice`, you may find a [`connectGatt` method with `opportunistic` parameter](https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/bluetooth/BluetoothDevice.java#2663). It has to be called using reflections, and the max target SDK is R, so following versions may not work. I need to check that. This method allows to create a weak connection, which does not prevent Android from disconnecting, when the last strong client disconnects. You may test it with nRF Connect 4.23.3, which still targets R. This type of connection is created when the app discovers a new connection, that hasn't been initiated by nRF Connect, and you'll see `opportunistic = true` in the logs.
A weak connection may also be created when a peripheral connects to your GATT server. If you will get a [callback](https://developer.android.com/reference/android/bluetooth/BluetoothGattServerCallback#onConnectionStateChange(android.bluetooth.BluetoothDevice,%20int,%20int)), that a device has connected, this connection should be weak. From this point, you may either connect to the same device as a client (which Ble Manager in this library si doing), or call `server.connect(device)` to let the system know, that you want the connection to be strong. When you are done, you may call `cancalConnection()` and this should weaken the connection. This is the intended behavior of those methods, I'm not sure how do they work in practice, I didn't try them on recent versions of Android, and actually I gave up on calling `server.connect()` in nRF Connect some time ago and I only rely on client connection.

georrge1994, 2021-12-01:
Hellow, @philips77 . Thank you for idea with `disconnect().enqueue()` - I already found it and it works =) Summary I am forced to use API a little wrong:

We have a GATT-server on android device. You know, that android provide dynamic mac address for every each new `startAdvertising`. Mac-address could be static only for bonded devices or if you have an own OS build with disable dynamic mac-address flag - we don't have then. Our clients are FTMS applications, there are a lot of and client's devices are too (includes IOS). I could not make stable and workable version with bonding: some devices work only once and then drop connection after 30  seconds with 133, another just doesn't connect after with bonding. I can not change client's app and use cache clearing for example (it help in my tests). In result I am forced to use Gatt-server without bonding. Our app work infinitely and I try to save mac address so long as possible -> I just don't disable advertisement when first client will connect. I'm needed option to disconnect another clients and I found follow solution:
 ```kotlin
            connect(device).enqueue()
            disconnect().enqueue()
```
I know looks not good, but it give me desired result. Maybe you can advice with bonding problem? Bonding idea like be better, but it make debugging mode infinity -> too much problems.

closed 2021-12-01 by georrge1994