nordicsemi/Android-BLE-Library #335: connect fails if initial bonding fails

cjurjiu, 2021-12-03:
Hi! Thanks for the library, really useful for us so far!

We have this bluetooth low-energy device which initiates bonding as soon as a connection is established. However, we don't need to bond to the device, and instead need to use the device without bonding.

We've noticed that calling `BleManager.connect()` however always fails if the initial bonding fails -> this makes it impossible for us to actually interact with the lock.

From looking through the code, I've noticed that if the bond sate goes from `BOND_BONDING` to `BOND_NONE`, service discovery is never even performed, so then connect fails in the end.

using `getServiceDiscoveryDelay` is not an option for us since that one only works if we can determine a delay long enough for bonding to already fail, but short enough for the bluetooth device to not time out our connection request. we can't change how the bluetooth device works, and noticed that we get disconnected from the gatt server immediately after the bonding request fails.

any way to work around this? ideally `BleManager.connect()` would connect to the gatt server & perform service discovery regardless of the bound state.

cjurjiu, 2022-01-30:
would appreciate some feedback/thoughts here..

philips77, 2022-02-02:
Let me look into the code and come back to you. I have to say, your use case is quite unique.

philips77, 2022-02-02:
Merged. I'll release it as 2.4.0 soon.

closed 2022-02-02 by philips77

cjurjiu, 2022-02-04:
excellent, thank you! I'll also recheck on Monday once I get to the office with our test BTLE device