nordicsemi/Android-BLE-Library #336: cancel useAutoConnect

cocomeneau, 2021-12-07:
Hi,
I would like to know if it's possible to change the status of `useAutoConnect` to false when the device is already connected. I've a device that's connecting with `useAutoConnect(true)` it's usefull when there is a link loss but I've some cases when I don't want my device to re-connect automatically. Is there a way to cancel this re-connection ?

philips77, 2021-12-07:
I didn't think of such use case.. Thank you for the feedback. I don't know how to easily get it done.

junebeans, 2021-12-18:
@cocomeneau  Just dropping my thought to make the above scenario available from a library consumer point.

When you monitor link loss to happen and once you call disconnect when the event is observer, GATT server will be closed and no longer re-connect to the peripherals.

philips77, 2022-02-02:
Thank you @junebeans. You're right. Disconnection closes the GATT client object.

closed 2022-02-02 by philips77