nordicsemi/Android-BLE-Library #344: disconnect() does not disconnect the device

sitannsu, 2022-01-10:
I am using below versions

 
    implementation 'no.nordicsemi.android:ble:2.2.4'
    implementation 'no.nordicsemi.android:ble-common:2.2.4'
    implementation 'no.nordicsemi.android:ble-livedata:2.2.4'

Using disconnect() method never diconnects device . I can see bluetooth symbol in device.

Is this a  existing issue?

VernonLee, 2022-01-26:
+1, huawei phone

philips77, 2022-02-02:
You didn't provide enough information to replicate the issue.
Are you using `disconnect().enqueue()`?
Could you share logs?
It may also be that another app is using the same connection. `disconnect()` just closes single client connection, but a different client may still be connected. A different client may be in another app, in your app, or even the system can be connected if it natively supports a service present on a device, e.g. HID. To be sure that the device disconnects you need to implement it on the device side and send a command from the phone.

closed 2022-02-02 by philips77