nordicsemi/Android-BLE-Library #345: Other chached devices wrongly get connected to phone when conecting to specific device usin MAC address

aamacaya, 2022-01-11:
I have the following problem:

First, as a starting point, the Bluetooth cache and data are completely deleted manually from the phone.
Then I have three equal devices (with different MAC addresses) all of them on advertising mode. 
I successfully connect to the first device. The other two devices are still in advertising mode.
I successfully disconnect from the first device.

All of the three devices are in advertising mode again.
I try to connect to the second device and I successfully connect to it. But strangely, the first device also gets connected to the phone, even though I'm using the MAC address to establish the connection. (The third device is still on advertising mode).
I try to disconnect from the devices. I successfully disconnect from the second device but the first one is still connected to the phone and never disconnects.

I reset the first device. All of the three devices are in advertising mode again.
I try to connect to the third device and I successfully connect to it. But the first and second devices also get connected.
I try to disconnect from the devices. I successfully disconnect from the third device but the first and second one are still connected to the phone and never disconnect.

I reset all the devices. All of the three devices are in advertising mode again.
Now every time I try to connect to any of the devices using the specific MAC address, all of the devices get connected to the phone, and only the device that I requested connection is able to disconnect.

The only solution to the problem is to manually delete the Bluetooth cache and data from the phone, but the problem will reappear as the devices are remembered by the phone.

I don't understand why other devices get connected even if the MAC address does not match. It seems to be a problem with the cache. I have defined shouldClearCacheWhenDisconnected() as true to try not to save any cache of the devices but the problem persists.

```java
        @Override
	protected boolean shouldClearCacheWhenDisconnected() {
		return true;
	}
``` 

How can I make sure that only the device (one device) with the desired MAC address gets connected and no other devices get connected to the phone? Is this a problem of the connection process or the disconnection process? Or is it a problem with the firmware of my devices?

Here are my connection/reconnection and disconnection functions:

Note that both the connection and disconnection Requests always return successfully.

```java
    public void connect(final DiscoveredBluetoothDevice target) {
        device = target.getDevice();
        savedMAC = device.getAddress();
        reconnect();
    }

    public void reconnect() {
        if (isConnected()) return;
        if (device == null) device = mBluetoothAdapter.getRemoteDevice(savedMAC);

        connectRequest = BleManager.connect(device)
                .retry(3, 100)
                .useAutoConnect(false);
        connectRequest.done(new SuccessCallback() {
            @Override
            public void onRequestCompleted(@NonNull BluetoothDevice device) {
                mConnectionListener.onDeviceConnected();
            }
        });
        connectRequest.fail(new FailCallback() {
            @Override
            public void onRequestFailed(@NonNull BluetoothDevice device, int status) {
                disconnect(true);      // If connection fails, restart the connection process (first forget device, then reconnect)
            }
        });
        connectRequest.then(new AfterCallback() {
            @Override
            public void onRequestFinished(@NonNull BluetoothDevice device) {
                connectRequest = null;
            }
        });
        connectRequest.enqueue();
    }

    public void disconnect(boolean reconnect) {
        if (connectRequest != null) {
            connectRequest.cancelPendingConnection();
        }
        if (BleManager.isConnected()) {
            disconnectRequest = BleManager.disconnect();
            disconnectRequest.done(new SuccessCallback() {
                @Override
                public void onRequestCompleted(@NonNull BluetoothDevice device) {
                    endDisconnect(reconnect);
                }
            });
            disconnectRequest.fail(new FailCallback() {
                @Override
                public void onRequestFailed(@NonNull BluetoothDevice device, int status) {
                    disconnect(reconnect);  // If disconnect fails, retry disconnecting.
                }
            });
            disconnectRequest.then(new AfterCallback() {
                @Override
                public void onRequestFinished(@NonNull BluetoothDevice device) {
                    disconnectRequest = null;
                }
            });
            disconnectRequest.enqueue();
        }
    }

    public void endDisconnect(boolean reconnect) {
        device = null;
        if (reconnect) reconnect();
        else {
            mConnectionListener.onDeviceDisconnected();
        }
    }

```

aamacaya, 2022-02-04:
New information:

It seems that the issue is related to the phone.

I was working with a Xiaomi Redmi 9. 

I tested the app on a Xiaomi 11 Lite and the issue does not happen.

Still, I'm interested to know if there is a way to solve te issue through the Android BLE Library

sylwester-zielinski, 2022-02-16:
The issue sounds strangely familiar. Are you using CompanionDeviceManager for scanning?
https://developer.android.com/reference/android/companion/CompanionDeviceManager

philips77, 2022-03-21:
I'm closing the issue due to inactivity. Feel free to reopen if needed.

closed 2022-03-21 by philips77