nordicsemi/Android-BLE-Library #346: Memory allocation due to logging

x2764tech, 2022-01-17:
I'm streaming packets from a device that sends a lot of data via notifications.
During memory profiling, I'm seeing that [`BleManagerHandler.gattCallback.onCharacteristicChanged`][1] allocates several strings for each packet. These seem to be soley for the purpose of logging (which my implementation then ignores)
Would it be possible for the library to check whether Debug logging is enabled *before* calling `ParserUtils.parse` and `log(level.Debug, ...)`, and thereby avoid these unnecessary string allocations?

[1]: https://github.com/NordicSemiconductor/Android-BLE-Library/blob/6d3c95f2fdb75485cdcb489e19b912e9173660a1/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L2287

philips77, 2022-01-18:
Yes, we observed it as well. I'll add option to disable logs, one optimize them in the next version.

x2764tech, 2022-01-20:
@philips77 Amazing. Thank you 🙏 
Do you have any idea of when the next release might be? More for managing expectations than anything else.

philips77, 2022-01-20:
I already implemented changes and doing some testing. I'll push the changes today and create a PR for review.

philips77, 2022-01-27:
Ok, that "today" became a week. I promise never to promise anything more!
But have a look at the PR and check if you need anything more.

PR #348 (2022-01-31): Option to filter logs by priority
This PR fixes #346.
The new method 
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/e55d46bf7459d9fb76c71908b861cb01a493f4c8/ble/src/main/java/no/nordicsemi/android/ble/utils/ILogger.java#L32-L37
in `BleManager` and `BleServerManager` allows to omit creating and logging logs on lower priorities. By default, logs from levels `Log.INFO` and higher will be logged, which is a change in default behavior. Previously all logs were produced, even if later ignored. 
To disable logging, return `Log.ERROR` or `Log.ASSERT` from those methods. To restore pr [... cut]
closed 2022-01-31 by philips77