nordicsemi/Android-BLE-Library #347: Reading encrypted characteristcs returns error 132 but bond state becomes BONDED

mecoFarid, 2022-01-26:
I have a simple setup as follows, where I am trying to bond with a host BL device by reading encrypted characteristics. This code sometimes succeeds but fails most of the time.

  ```
  class BleClient (context: Context): BleManager(context){
    fun connectAdapter(): Flow<ConnectionState> = stateAsFlow().also {
      connect([BluetoothDevice])
        .useAutoConnect(true)
        .enqueue()
    }

    fun bond(): Flow<BondState> =
      bondingStateAsFlow().also {
        readEncryptedCharacteristic()
      }

    private fun readEncryptedCharacteristic() {
      super.readCharacteristic([ENCRYPTED_CHARACTERISTIC]).enqueue()
    }
  }
```

**Usage**
```
  val bleClient = BleClient(context)
  // Wait for adapter connection to be ready
  bleClient.connectAdapter().first { it == ConnectionState.Ready }
  // Wait until bonded
  val bonded  =  bleClient.bond().first { it = BondState.Bonded }
  Log.i(TAG, "Bond state: $bonded")
```

When `bleClient.bond().first { it = BondState.Bonded }` is executed both host and the client displays a pairing dialog, once I click on "confirm" on both machines. The logcat shows the following logs: How is that even possible that reading encrypted characteristics fails but bond state becomes `Bonded`?

> com.example.app E/BleManager: onCharacteristicRead error 132
> com.example.app I/BleClient: Bond state: : no.nordicsemi.android.ble.ktx.state.BondState$Bonded@d9512fe

**Note:** `error 132` means GhattDevice is busy

philips77, 2022-02-16:
> How is that even possible that reading encrypted characteristics fails but bond state becomes Bonded?

Until around Android 8 the system wasn't automatically repeating a request in case of a lack of authentication. The library tries to do this for you. Have a look at 
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/b5e9580d2e4796934f1d5f0e963dc9cc2c917b98/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L375-L396

What phone are you testing on, and what is the Android version?
What authentication method do you have on the encrypted characteristic? 
Did you make any progress with the issue? Sorry for looking so late into it.

philips77, 2022-03-21:
Any updates on this issue?

mecoFarid, 2022-03-26:
@philips77 This is now working but, honestly, I have played around with so many parameters that I don't know exactly which setting helped. Just as an example I, definitely, know I'm passing different values to these 2 parameters than I did before. Previously I was passing no flag or `true`, now passing true to both of them. 

`ScanSettings.Builder.setUseHardwareBatchingIfSupported(true)`
`ConnectRequest.useAutoConnect(true)`

Can we keep this open until I get some free time to play around to reproduce the issue so we (or you) can resolve it once for all?