nordicsemi/Android-BLE-Library #354: On the problem of incomplete received data

phoebewow78, 2022-02-08:
D/BluetoothGatt: writeCharacteristic() - uuid: 0000f1f1-0000-1000-8000-00805f9b34fb
D/BluetoothGatt: onCharacteristicWrite() - Device=50:33:8B:**:**:** handle=18 Status=0
D/BluetoothGatt: onNotify() - Device=50:33:8B:**:**:** handle=21
D/BluetoothGatt: onNotify() - Device=50:33:8B:**:**:** handle=21

After sending the data, the length of the correctly received data is 24 bytes, but in fact, the data received by setnotificationcallback is only 20 bytes, and another 4 bytes disappear.

I see the log above. It should be received in two packets, but your Bluetooth library only returned 20 bytes to me.

I don't know if the interval between the two packets of data is a problem. Is there any way to solve it?

philips77, 2022-02-08:
Hello,
Did you set a merger using `.merge(..)` in the notification callback?
You may also increase MTU, if your device allows, so that returned data is sent in a single notification.

philips77, 2022-02-08:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/8fc0fcddbaa551826305cf214f8f9660b65e9732/ble/src/main/java/no/nordicsemi/android/ble/ValueChangedCallback.java#L110-L122

phoebewow78, 2022-02-08:
ok， i try it. thanks.

phoebewow78, 2022-02-08:
Thank you very much for your reply!!!

Is there any relevant demonstration code?

I need to use it in datareceivedcallback.

phoebewow78, 2022-02-08:
I have found a way to use it and can receive data correctly. Thank you.

closed 2022-02-08 by philips77