nordicsemi/Android-BLE-Library #356: BleManager.isConnected() returns true even if peer device's Bluetooth is turned off

mecoFarid, 2022-02-10:
Steps to reproduce:
1. Connect to a device
2. Turn off Bluetooth in the peer device
3. Call BleManager.isConnected()

The method will still return `true` even if the peer device has its Bluetooth turned off.

philips77, 2022-02-10:
It takes *supervision timeout* milliseconds for the connections state to be passed to the apps. This is usually 4 or more seconds, and is set together with other connection parameters: connection interval and slave latency. After few seconds you should get disconnection event and from that moment this method should return false.

philips77, 2022-02-10:
If you still experience this issue even after supervision timeout passed, please reopen the issue. Also, mention which lib version are you using.

closed 2022-02-10 by philips77

mecoFarid, 2022-02-10:
@philips77 I edited my answer with more details.  I waited for more than 3 mins. Should I open another issue?

philips77, 2022-02-10:
What device are you using as the remote device? Did you try turning it off instead of just turning Bluetooth off? On some devices, e.g. on iOS, disabling Bluetooth in Control Panel just blocks new connections, but doesn't actually turn off Bluetooth for existing devices. Try turning the device off or move away from it.

mecoFarid, 2022-02-10:
The peer device is iOS. Let me check it and will get back to you. This might be the answer for the #357 too.

mecoFarid, 2022-02-10:
And yes you are right, turning off the iOS device notifies disconnection. I appreciate your quick reply. Your help and reply is very much appreciated